libplctag/libplctag #466: Support for Omron NX/NJ variable listing

MountainKing91, 2024-06-15:
I have noticed that some tools/libraries are able to "discover" Ethernet/IP published tags in Omron NX/NJ controllers, while libplctag is not. I understand this is because of differences in the protocol implementation by different vendors.

I'd like to share the wireshark captures of listing the tags available in NX1P2 controllers when there are no global variables published by the programmer. In such situation, the tags listing returns all the controller **system variables** (they all start with the reserved starting character "_").
[omron_get_variable_list.zip](https://github.com/user-attachments/files/15846609/omron_get_variable_list.zip)

This is the variable list:
[nx1p2_variable_list.txt](https://github.com/user-attachments/files/15846613/nx1p2_variable_list.txt)

I hope this can help to implement the tag listing for omron devices.

kyle-github, 2024-06-15:
Thanks for the captures!   I have been examining the Aphytcomm project and it shows that Omron implements almost everything either slightly differently or very differently than Rockwell.

The challenge for libplctag is that I implemented all of the EIP/CIP PLC support in one set of code with some switches to behave a bit differently based on the PLC type.  Unfortunately, Omron is just too different.  I am currently working on refactoring out the common parts of the code so that I can have much better control over just how each PLC is handled.   And this is going to take some time.  I do not have an ETA for the work at this point.

MountainKing91, 2024-06-15:
No problem, I'm just happy if I can be of some help. Feel free to reach out if you need captures of something specific or you need some test.

kyle-github, 2024-06-16:
Interesting.  Rockwell uses the class 6A as the data table.  It contains information on where tag data is stored.  The tags themselves are in 6B though sometimes that is embedded in another class like with program tags.

Do you have a capture where Aphyt is listing the regular tags?

I hope that there is a better way to get tag information than this.   I have a PLC with several hundred tags.  I can't imagine how long that would take doing them one at a time like that!

MountainKing91, 2024-06-16:
Ok. By _regular tags_ i assume you mean user defined global variables. These are the only tags that - is set with the "Network publish" property to "Publish Only", "Input" or "Output" - can be R/W with ethernet/ip.

This is my global variables table:
[image]

User defined data types:
[image]
[image]

Apthyt code:
```
import time
from aphyt import omron

if __name__ == '__main__':
    
    with omron.NSeries('192.168.250.1') as eip_conn:

        var_list = eip_conn._get_variable_list()
        for v in var_list:
            print(v)
```

captures:
[captures_tags.zip](https://github.com/user-attachments/files/15857271/captures_tags.zip)

I actually don't know if there is another way of listing the tags with aphyt (i.e. listing ONLY the user defined tags, not the system ones), maybe I could try to see how other libraries are doing?

kyle-github, 2024-06-19:
I think the system tags start with an underscore.

MountainKing91, 2024-06-19:
Yes (opening comment)

kyle-github, 2024-06-19:
It is strange.  The attributes retrieved for each tag object instance are really minimal.  I see nothing that would suggest the array dimensions for instance.

The pattern I see here is that Aphyt gets all attributes on the CIP class 6A.  This includes four fields which all have the value B8.   If this is anything like Rockwell, one of them is probably the number of class instances and the other is the highest ID.  In this case, those are the same.

Then the library just iterates over the instance IDs from 1 to B8 and gets all attributes on those.  The attributes are not defined but there is clearly string data.  When the total number of bytes in the string is odd (counting the count byte) then there is padding.  Only one byte though.  So I think the next field is probably a 16-bit number.  The value of which happens to match the instance ID of the object. 

The first found bytes of the response are interesting.   They look very much like a class/instance pair.   Only the class is 6B, not 6A.  The instance ID is the same.  Rockwell uses class 6B for the symbol/tag instances.   

This helps a lot.  I wish I had an Omron PLC I could poke at.  Now that I've seen this I have a better idea of what Aphyt is doing.

MountainKing91, 2024-06-19:
If you'd like to have captures of something specific, again just ask and I'll do my best.

kyle-github, 2024-06-19:
I'm going to write some small programs that send off some of these same queries, but then decode what I can.  Can you compile and use C?   If not, I'll have to lean on Tim to translate to C#.

MountainKing91, 2024-06-19:
I am not very familiar with C, but I could try to install a compiler and go from there? I guess it's worth trying, you could always ask Tim later. Let's see.

kyle-github, 2024-06-19:
I may try with Python to test my luck :-)

MountainKing91, 2024-06-19:
All right, either way, I'm available.

timyhac, 2024-06-19:
If you're using Visual Studio on Windows, compiling libplctag just requires you to install the "Desktop Development with C++" workload ad it is quite straightforward to compile the C programs.

MountainKing91, 2024-06-19:
That's my setup, VS 2019 on Windows 10. 
You mentioned python, I am also equipped with that.
What should I do exactly?

kyle-github, 2024-06-20:
```c
#include <stdbool.h>
#include <stdint.h>
#include <inttypes.h>
#include <stdio.h>
#include <stdlib.h>
#include "../lib/libplctag.h"
#include "utils.h"

/*
 * This is a partial example of how we might do Omron tag listing.
 */

#define REQUIRED_VERSION 2,4,0

/***************** CHANGE THE IP ADDRESS TO YOUR PLC's ADDRESS ******************/
#define TAG_STRING "protocol=ab-eip&gateway=10.206.1.40&path=1,0&plc=ControlLogix&name=@raw"
/******************************************************************************************/
/* yes, I think we have fake that it is a ControlLogix for now. */

#define DATA_TIMEOUT 5000

enum {
    Get_Attributes_All = 0x01,
    Get_Attribute_List = 0x03,
    Get_Attribute_Single = 0x0E,
    Find_Next_Object_Instance = 0x11,
    Symbolic_Translation = 0x4B,
    Read_Tag_Fragmented = 0x52,
    Get_All_Tags_Attributes = 0x55,
} cip_commands;

/*

Sample read command: Read Test

Read Tag
52 0a 91 12 54 65 73 74 4d 61 6e 79 42 4f 4f 4c 46 69 65 6c 64 73 01 00 00 00 00 00

Get_Attributes_All
01 0a 91 12 54 65 73 74 4d 61 6e 79 42 4f 4f 4c 46 69 65 6c 64 73

Get_Attribute_List
03 0a 91 12 54 65 73 74 4d 61 6e 79 42 4f 4f 4c 46 69 65 6c 64 73 01 00 01 00

Get_Attribute_Single
0e 0a 91 12 54 65 73 74 4d 61 6e 79 42 4f 4f 4c 46 69 65 6c 64 73 30 01

Find_Next_Object_Instance
11 0a 91 12 54 65 73 74 4d 61 6e 79 42 4f 4f 4c 46 69 65 6c 64 73
*/

static int set_tag_data(int32_t tag, uint8_t *data, size_t raw_data_size)
{
    int rc = PLCTAG_STATUS_OK;
    int data_size = (int)(unsigned int)raw_data_size;

    rc = plc_tag_set_size(tag, data_size);
    if(rc < 0) {
        printf( "\nERROR: Unable to set the payload size on the tag %s!\n", plc_tag_decode_error(rc));
        return rc;
    }

    rc = PLCTAG_STATUS_OK;

    for(int i=0; i < data_size && rc == PLCTAG_STATUS_OK; i++) {
        // printf("*** Setting index %d to %x.\n", i, (int)(unsigned int)get_attribute_list[i]);
        rc = plc_tag_set_uint8(tag, i, data[i]);

        if(rc != PLCTAG_STATUS_OK) {
[... 232 lines cut ...]
                break;
            }

            if(rc == PLCTAG_STATUS_OK) {
                printf("Instance 0x%02x: \"%s\".\n", id, tag_name);
            } else {
                printf("Instance 0x%02x seems to be unused.\n", id);
            }
        }
    } while(0);

    if(tag > 0) {
        plc_tag_destroy(tag);
    }

    printf("\n\n********* Complete *********!\n");

    return 0;
}

```

I do not have an Omron PLC so I cannot test this. It compiles!

I got a little carried away.  I had created a little test program that let me really quickly fire off CIP commands to test various things.  I just repurposed it a bit.

kyle-github, 2024-06-20:
It looks like Aphyt does this listing to get the tag names and then done Get_Attributes_All on the tag name itself (encoded as an extended symbolic segment).

kyle-github, 2024-06-20:
I am sure this would be nicer in C# or Python.

timyhac, 2024-06-20:
@kyle-github - `@raw` is paying dividends, good foresight!
@MountainKing91 - If you do want to convert to C# and get stuck using the main .NET wrapper, you should be able to use the NativeImport package and it should be virtually a copy-paste of the C code. See example: https://github.com/libplctag/libplctag.NET/blob/master/src/Examples/CSharp%20NativeImport/NativeImportExample.cs

kyle-github, 2024-06-20:
I probably will need to work on other things.  I am trying to go on vacation in a week :-)

MountainKing91, 2024-06-20:
> If you're using Visual Studio on Windows, compiling libplctag just requires you to install the "Desktop Development with C++" workload ad it is quite straightforward to compile the C programs.

@kyle-github I'm installing right now, it's going to take a while (6gb, slow connection), once it's done I'll try to compile and run. If something goes wrong I'll consider porting to C#.

MountainKing91, 2024-06-20:
@kyle-github 
Installed. Is this what I need to do? https://github.com/libplctag/libplctag/blob/release/BUILD.md
[image]

timyhac, 2024-06-20:
Yes - but of course you'll additionally need to save the new source code Kyle listed in the comment above and add it to the [CMakeList](https://github.com/libplctag/libplctag/blob/release/CMakeLists.txt#L496)

MountainKing91, 2024-06-20:
All right, this should be ok:
[image]

I also changed the IP address but not the plc type as indicated
[image]

If I run the code I get return code 0x0, which is great I guess, but I can't any output

MountainKing91, 2024-06-20:
I placed a breakpoint at the end before returning and in the console I read this:
```
Starting with library version 2.6.0.

        Sending data:  01 03 20 6a 25 00 00 00
        Received response data:  81 00 00 00 00 01 cd 00 cd 00 ff ff 00 bc 87 77 53 00 00 00 cd 00 00 00 cd 00 00 00 ad 00 00 00 20 00 00 00
        Sending data:  01 03 20 6a 25 00 01 00
        Received response data:  81 00 00 00 6b 00 01 00 07 5f 4d 43 5f 47 52 50 01 00 00 00
ERROR: CIP command failed on remote PLC with error code 0!ERROR PLCTAG_ERR_REMOTE_ERR: Could not run Get_Attribute_All on tag instance!

********* Complete *********!
```

kyle-github, 2024-06-20:
About line 316, change the debug level to PLCTAG_DEBUG_DETAIL.  That should dump out the logs.

kyle-github, 2024-06-20:
Here's the file: 
[list_tags_omron.c.zip](https://github.com/user-attachments/files/15915048/list_tags_omron.c.zip)

kyle-github, 2024-06-20:
I checked it does compile.

MountainKing91, 2024-06-20:
here's the dump:
[details.txt](https://github.com/user-attachments/files/15915230/details.txt)

MountainKing91, 2024-06-20:
> The Omron refused and sent back 1996 bytes
This is the max packet right? I found out of this limit yesterday evening browsing the issues after I couldn't read an array of 16 400 bytes udts - I had to read part of the array separately and join them afterwards.

kyle-github, 2024-06-20:
Well, it depends on who you ask.  I just dropped the Omron default to 1994 because that is what is documented in the Omron docs according to another library user who has been doing a lot of the Omron changes lately.  That change has not landed yet.  Soon though.

This timeout thing is weird.  That means maybe my timeout settings are wonky and it just happens to work with my Rockwell systems.   The second attempt is sent right after the first error is received but there is no reply for a long time.  Then we get the timeout.

MountainKing91, 2024-06-20:
updated the comment

kyle-github, 2024-06-20:
Another timeout, but no failure before...  This is definitely not what I expected.   Let me look closely at the timeout fields to make sure I did not accidentally change something.

kyle-github, 2024-06-20:
[list_tags_omron.c.zip](https://github.com/user-attachments/files/15915875/list_tags_omron.c.zip)

Another try with different timeout parameters.  Maybe this will help?

kyle-github, 2024-06-20:
You are still going to need to change the IP.

MountainKing91, 2024-06-20:
[details.txt](https://github.com/user-attachments/files/15915915/details.txt)

kyle-github, 2024-06-20:
It was 150 ms (plus or minus a few) on the last attempt too, so I don't think this is a random thing.

kyle-github, 2024-06-20:
Same... Oh, wait, try setting the path to 1,0.   I just realized that the path was still set for my ControlLogix.

kyle-github, 2024-06-20:
Also, 150ms again.

MountainKing91, 2024-06-20:
I did not notice that either, should have - now it's 1,0
[details.txt](https://github.com/user-attachments/files/15916130/details.txt)

kyle-github, 2024-06-20:
So close.  Apparently the code is deciding that error status 0 means a bad thing.  It doesn't.  Let me see what is going on.  It can see that it got the number of instances and max ID.  Looks like you have 238 tag instances (most of which are probably system tags).

MountainKing91, 2024-06-20:
The system tags should be 173

kyle-github, 2024-06-20:
Ah... Apparently I cannot count.  I was forgetting that the 4-byte CIP response header was still on the response since these are raw tags.  I am changing the code to point to the right byte offsets.

kyle-github, 2024-06-20:
[list_tags_omron.c.zip](https://github.com/user-attachments/files/15916449/list_tags_omron.c.zip)

Try this.  Same drill: change the IP and the path.

kyle-github, 2024-06-20:
Missed another offset.

kyle-github, 2024-06-20:
[list_tags_omron.c.zip](https://github.com/user-attachments/files/15916499/list_tags_omron.c.zip)

Let's try that...   There are some more outputs too.  Look for "INFO:" in the logs.

MountainKing91, 2024-06-20:
[details.txt](https://github.com/user-attachments/files/15916528/details.txt)

kyle-github, 2024-06-20:
Looks like it worked!
 
You can turn off the debugging to see the useful output.

MountainKing91, 2024-06-20:
Awesome! I recognize the system tags and also the ones I created. Next steps? =)

I should find a way to get you an omron nx!!!

kyle-github, 2024-06-20:
Figure out what Aphyt is doing to get the rest of the information.  There are a number of different classes that are hit but the entry point is not as clear.  I do not know if just the tag name is used or if the instance ID is used.

MountainKing91, 2024-06-20:
All right, I'm available if needed!

kyle-github, 2024-06-20:
I need to do some planning.  I am going to be out for a long vacation in less than a week so I want to make sure that whatever I do is stable before then.

kyle-github, 2024-06-21:
Here's an update.  This should get some of the tag metadata.  It gets the tag type.  It gets some other ID that Aphyt gets for the type info object instance and possibly a size.  Maybe the size is only valid for strings?

[list_tags_omron.c.zip](https://github.com/user-attachments/files/15921418/list_tags_omron.c.zip)

kyle-github, 2024-06-21:
It might be feasible to get the metadata about a tag when we get the tag.  It would pretty much require splitting Omron out because there are so many differences.

MountainKing91, 2024-06-21:
Looks like we have an error this time
[details.txt](https://github.com/user-attachments/files/15925081/details.txt)

MountainKing91, 2024-06-21:
Actually I tried again the new code and it worked fine, I am not sure if the first time I tried there was some other issue (cable?)
`// plc_tag_set_debug_level(PLCTAG_DEBUG_DETAIL);`
[no_details.txt](https://github.com/user-attachments/files/15926486/no_details.txt)
`plc_tag_set_debug_level(PLCTAG_DEBUG_DETAIL);`
[details.txt](https://github.com/user-attachments/files/15926487/details.txt)

I see that it's getting metadata as well

kyle-github, 2024-06-21:
Hmm, it looks like I have the offsets wrong. I am pretty sure they are not all type 0x02c1.   I think the type should be one byte.   And the other data looks off.  That said, in Aphyt, the size part only is accessed when the data type is a string, so perhaps that is not valid otherwise?  If these really are mostly BOOL types, then 2 bytes is correct (which is what I think I see in the payload data.

MountainKing91, 2024-06-21:
I can provide an "exploration" of the various tags, I did it with Aphyt, some are complex datatypes actually. You can probably check faster on the manual, chapter 3-2:
[image]

kyle-github, 2024-06-21:
Ignore the printing error that makes the types look like two bytes.  The 02 at the end of each type was supposed to be a format attribute for printf.

There are a couple of findings.  

- unlike Rockwell, it doesn't look like Omron uses the short versions of the UDT and array types.  These are types a0 (UDT) and a1 (array).  Instead they use a2 and a3 which are the "full fat" fully described versions.
- In the tag data for arrays, at least, there is a suspicious data type right after the type for the array.  This corresponds to the size as well.
- I think the size, for arrays, might be the element size.  It varies much more for UDTs.

Here's some arrays:

```
grep -B 1 -A 2 "Tag data type a302" details.txt | grep -v "Tag name is" 
        Received response data:  81 00 00 00 02 00 00 00 a3 c1 01 00 c0 00 00 00 00 00 00 00 00 00 00 00 00 04 02 00 00 00 00 00 01 00 00 00INFO: Tag "_EC_EntrySlavTbl" has attributes:
        Tag data type a302
        Tag data type instance ID 00c0
        Tag size bytes 2
```
Above we have an array.  Immediately after the A3 type byte is another byte that is C1, bit/BOOL.  Those are 16 bits in memory if not packed and still take up chunks of 16 bits when packed.  So the size here makes sense if this really is an array of BOOL and it is relatively short.  

```
 81 00 00 00 02 00 00 00 a3 d2 02 00 04 00 00 00 0a 00 00 00 00 00 00 00 00 00 00 00 00 84 02 00 00 00 00 00 00 00 00 00 00 00 00 00INFO: Tag "_CJB_UnitErrSta" has attributes:
        Tag data type a302
        Tag data type instance ID 0004
        Tag size bytes 2
```
Here the array might be composed of type D2 (right after the A3). Type D2 is a 16-bit bit string.  Two bytes.  Notice the tag size?  Might be the tag size. Might be the element size.

```
 81 00 00 00 02 00 00 00 a3 d2 01 00 08 00 00 00 00 00 00 00 00 00 00 00 00 04 02 01 00 00 00 00 00 00 00 00INFO: Tag "_MC_GRP_ErrSta" has attributes:
        Tag data type a302
        Tag data type instance ID 0008
        Tag size bytes 2
```

Type d2 again.  Same size too.

```
 00 00 00 06 00 00 00 a3 a2 01 00 02 00 00 00 c3 00 00 00 00 00 00 00 00 84 02 00 c3 00 00 00 01 00 00 00INFO: Tag "_PLC_OptBoardSta" has attributes:
        Tag data type a302
        Tag data type instance ID 0002
        Tag size bytes 6
```

Here's something different the type after the array type byte is A2.  That's a UDT type.   And the size is 6 not one of the 1, 2, 4 or 8 byte sizes of the base types.  Is that the size of the whole thing or the element?

Here's two UDTs:

```
grep -B 1 -A 2 "Tag data type a202" details.txt | grep -v "Tag name is" 
 81 00 00 00 64 00 00 00 a2 00 00 00 ae 00 00 00 00 00 00 00 00 04 02 01 ae 00 00 00INFO: Tag "_MC_COM" has attributes:
        Tag data type a202
        Tag data type instance ID 00ae
        Tag size bytes 100
--
 81 00 00 00 48 00 00 00 a2 00 00 00 bd 00 00 00 00 00 00 00 00 04 00 00 bd 00 00 00INFO: Tag "_Card1BkupCmd" has attributes:
        Tag data type a202
        Tag data type instance ID 00bd
        Tag size bytes 72
```
These have sizes that look like they might be the whole UDT.  Notice the three bytes after the A2 byte.  They are all zero for UDTs.  At least these UDTs.

I think I need to implement another level of query to get the type info.  I'll code that up.

kyle-github, 2024-06-21:
There is some guesswork on the data here.   Can you please run this @MountainKing91?  You'll need the usual fixes.

[list_tags_micro8x0.c.zip](https://github.com/user-attachments/files/15928987/list_tags_micro8x0.c.zip)

If this does not crash, I will check it into the prerelease branch so that we have it archived.

This code should (!) pull a lot more information about arrays.  If I get a chance I'll try to prototype the UDT part as well.  All this comes from the same CIP class we were hitting yesterday.  There is some useful stuff in there, but it is not complete.   

I will be going on a long vacation this coming Wednesday, so I am running out of time to keep moving this forward for a few weeks.  I will be back for a week or so in late July and then gone again until about mid August.

MountainKing91, 2024-06-21:
> You'll need the usual fixes.

Here?
[image]

[149 more replies]