libplctag/libplctag #479: Unable to read from offsets in Output/Input datafiles on MicroLogix

CodingJWE, 2024-07-29:
I am having trouble reading singular bit values or the entire byte from anything other than the first offset in the input/output data files on a MicroLogix plc. An example of what I am attempting to read can be seen below:

[image]

Currently, I am only able to access I:0.0, is there a way to access the reset of the offsets? For reference, I am using the following tag names:
- I1:0.1 (does not work)
- I1:0/16 (does not work)
- I1:0 (works, but only for the first offset)

CodingJWE, 2024-07-29:
I:0.1 and I:0 also do not work

kyle-github, 2024-08-21:
See issue #481.  I think I fixed the requirement for the data file numbers.   Still not sure what is happening with the offsets though.   Input and Output are not treated quite the same way as "normal" data files in the PLC.  

The code in the library does not handle a number after a ".".   It is looking for a mnemonic (.acc etc.) and translates that into a sub-element number.  

It looks like the code needs to be changed to handle a number after a "." and to use that for the sub-element number.   

I think "I1:0" is working by accident.   It encoded the sub-element as 0 when there is no information about it in the tag string.   In this case there is not, so it defaults to zero.  That is equivalent to I1:0.0.

Let me fiddle with this a bit.  I think I can fix this.

commit c2218ddd (2024-08-21): Added fix for omitted data file number on I or O files. Added support for a numeric subelement.  Fix #479. (#483)
commit 7f99a7c3 (2024-08-21): Merge prerelease into release.  Fix #479. (#484)

* Added guard checks for not implemented vtable functions.   Several were missing!

* Added deprecation warning.

* Added support for raw tags.

* Fix up how printable characters are identified.  ASCII is smarter than I thought.

* Make file number optional if input or output file.  Fix #481.

* Added fix for omitted data file number on I or O files. Added support for a numeric subelement.  Fix #479. (#483)
commit ee3e60da (2024-08-21): Merge prerelease into release, bump version for tagging. (#485)

* Added guard checks for not implemented vtable functions.   Several were missing!

* Added deprecation warning.

* Added support for raw tags.

* Fix up how printable characters are identified.  ASCII is smarter than I thought.

* Make file number optional if input or output file.  Fix #481.

* Added fix for omitted data file number on I or O files. Added support for a numeric subelement.  Fix #479. (#483)

* Bumping version for release tag.
commit c55bc587 (2024-09-03): Prerelease merge into release (#491)

* Make file number optional if input or output file.  Fix #481.

* Added fix for omitted data file number on I or O files. Added support for a numeric subelement.  Fix #479. (#483)

* Fix #486. (#487).   String size change problem.   Fixed with #488.

* Added atomic flag and checks to aid in making all sessions terminate when the library is shutting down. Fix #450.

* plc_tag_set_string() feature additions (#488).   Setting string can change tag data buffer size.  And added flag, allow_field_resize, to suppress change.

* Added msp response size check befo [... cut]
closed 2024-08-21 by kyle-github