libplctag/libplctag #486: Unable to write value of STRING type for Omron NJ PLC using lib version (v2.6.1 )

ahmad9918, 2024-08-21:
Hello,

I am trying to write the value of string type for Omron NJ PLC which does not work. Read is working and I am able to fetch the value from the PLC and get the string using plc_tag_get_string() function. But when I try to write the value for the same string variable, it is throwing error and timeout is observed in the log "Timeout waiting for tags to finish writing! Error found: PLCTAG_ERR_TIMEOUT".

I am using below string attribute:
str_is_counted=1
str_count_word_bytes=2
str_is_zero_terminated=1
str_is_fixed_length=0
str_pad_bytes=0

Also I am setting the size of the buffer using plc_tag_set_size() function, more than the size of the string which is present in the PLC. Buffer size is more than the string size to hold the value. Further I am trying to set the string value and then try to write the value. It gives PLCTAG_ERR_TIMEOUT error and wireshark log is also attached. Failure is observed in below code snippet attached:
[image]

Wireshark Snippet:
[image]
[image]

Please share your inputs or suggestions. Thanks

kyle-github, 2024-08-21:
Thanks for using libplctag.   

I think this is related to other issues such as [#359](https://github.com/libplctag/libplctag.NET/issues/359) in the .NET wrapper repository.

Based on the Wireshark results, the PLC is definitely saying that there is too much data.  0x15 is the specific CIP error for "too much data".  

It appears that this is true.  The count word is 0x0005 (5 characters) but you are sending 6 characters.  I think Omron uses standard CIP strings which have a 2-byte count word.  I do not know what the maximum capacity is.  Probably 82 characters if it matches Rockwell.

I would try the settings:

- str_is_counted=1
- str_count_word_bytes=2
- str_is_zero_terminated=**0**  <-- changed
- str_is_fixed_length=0
- str_pad_bytes=0
- **str_max_capacity=82**  <-- new

I put the changes in bold.

Then set the tag buffer to the length of the string plus 2 bytes (for the count word).   In this case the size would be 7.

In the future please paste as much as possible in text rather than screen shots.  Thanks.  Also, you can [turn on debugging](https://github.com/libplctag/libplctag/wiki/API#debugging) in the library.  Use debug level DETAIL (4).  That will dump all the packets and most of the internal logic.   Capture that in a log file and attach it to the issue.  That way I can see everything the library is doing, not just the packet captures.

Let me know if this helps.

ahmad9918, 2024-08-21:
Hello Kyle,

Thanks for the quick response and the explanation given.
I have tried to test with above combination you have given in bold but still it could not write the value.
I have enabled debug level => PLCTAG_DEBUG_DETAIL and captured the logs in text file. It is attached for your reference.

[StringWriteIssue.txt](https://github.com/user-attachments/files/16692781/StringWriteIssue.txt)

kyle-github, 2024-08-21:
Hmm, I think you might have found a bug.  The code is using the max capacity to determine the amount of space required.  But it should not do that unless `str_is_fixed_length` is set to 1.  If that is zero, it should be using the actual string length.

Let me see if I can fix this.  I do not have an Omron PLC and all mine use the full amount of space when writing strings...

ahmad9918, 2024-08-21:
Thanks for the quick analysis.
I can help in test with Omron PLC after the fix. Let me know the branch details once the fix is ready so that I can build the source and verify if the fix is working or not.

kyle-github, 2024-08-21:
The prerelease branch has a proposed fix.  There are some changes here:

- plc_tag_set_string() really changed.  It now allocates more buffer space if you pass it a string larger than the buffer.   
- I fixed a couple of problems that showed up when plc_tag_set_string() and plc_tag_get_string() failed early.   

TODO: make sure that the buffer space is correct when the string is smaller.  That could be a problem.

ahmad9918, 2024-08-22:
Hello Kyle,

Thanks for providing quick fix.

I have verified string write issue with Omron NJ PLC, now write is working as expected. Tried with larger string, plc_tag_set_string() function was able to set the buffer size to the required length. Also, I have tried passing smaller string which fails if size is not set in the code. This point you have already mentioned in **TODO** list.

Thanks for the quick support 👍

kyle-github, 2024-08-22:
Having the tag resize is a bit of a problem.   If the string is in a UDT or an array of other strings, this will be the wrong action.  Resizing only works if the string is the only thing in the tag.

I may just change it anyway so that it works for the common case.

kyle-github, 2024-08-22:
I think I have fixed setting string shorter too.  I had to change some things.  If you have a string that is not a fixed size, the plc_tag_set_string() function will expand and contract the tag data buffer at the area of the string to accommodate the size change.  Data before the string is left exactly where it was and data after is moved but not changed.

If you get a chance, could you please test this from the prerelease branch?

ahmad9918, 2024-08-22:
Hello Kyle,

I have tested by passing shorter string without setting the size of the buffer, plc_tag_set_string() function is able to calculate the length of the string and set the buffer size to the required length. Write operation is getting success with shorter as well as longer strings.

Do let me know if you need detailed debug log for reference. Thanks

kyle-github, 2024-08-22:
Thanks for the testing!   

I think we can close this issue.  I'll make sure that this change gets into the next release.

commit 9770c4d0 (2024-08-21): Fix #486. (#487)

* updated version for this patch.

* Not sure what was going on here, but this test would never have worked.  Fixed.

* Added static initializer.

* Changed logging to DEBUG_DETAIL.  Might need to back that out.

* Added atomic flag and checks to aid in making all sessions terminate when the library is shutting down.

* Two sets of fixes: 1) added rc_dec(tag) where it was missing which led to tag leakage and some threads left running.  2) Rewrote plc_tag_set_string.  Hopefully it is right this time.  It now allocates more tag buffer space when there isn't enough.
commit c55bc587 (2024-09-03): Prerelease merge into release (#491)

* Make file number optional if input or output file.  Fix #481.

* Added fix for omitted data file number on I or O files. Added support for a numeric subelement.  Fix #479. (#483)

* Fix #486. (#487).   String size change problem.   Fixed with #488.

* Added atomic flag and checks to aid in making all sessions terminate when the library is shutting down. Fix #450.

* plc_tag_set_string() feature additions (#488).   Setting string can change tag data buffer size.  And added flag, allow_field_resize, to suppress change.

* Added msp response size check befo [... cut]
closed 2024-08-22 by kyle-github