libplctag/libplctag #493: How to report a security vulnerability

zenozom, 2024-10-07:
Hello,

I have two potential security vulnerabilities that I would like to report.

Before proceeding, I wanted to ask if opening a GitHub issue is the appropriate and preferred way to disclose security-related issues. Alternatively, if there is a more secure way that you recommend for handling security vulnerabilities, please let me know.

Best regards,
Gabriele.

kyle-github, 2024-10-07:
You can contact me directly. My email is in every source file.  Thanks for looking at this!   How are you finding these vulns?

zenozom, 2024-10-08:
Thanks for the response! The vulnerabilities have been found by fuzz testing the target functions with AFL++ fuzzer.

kyle-github, 2024-10-08:
Got the data.   That is a good catch!

kyle-github, 2024-10-28:
I have been going through the code to see if there are other areas like the ones you found and there appear to be other areas where the same kind of vulnerability exists.   This is taking a bit longer than I thought.

kyle-github, 2024-10-29:
Phwew.   By code inspection I can see a bunch of places that look questionable.   I will need to build a packet validation framework or library as well.   Sigh.   Thanks again for finding this.

zenozom, 2024-11-19:
Hi @kyle-github do you have any feedback about the reported vulnerabilities?
Besides the reported issues, are there other places where this vulnerable pattern occurs?

kyle-github, 2024-11-19:
Not yet.  I am dealing with some family health issues.   

The same kind of check is done in multiple places.  I _think_ it is being done correctly at the EIP level, but I need to make sure that the length is clamped everywhere that it could be written to or read into.  I also need to check that the multi-packet builder (when sending, not receiving) is also checking.  I _think_ I am but I need to make sure.

zenozom, 2024-12-10:
Hi @kyle-github,

I hope things are getting better on your end. I just wanted to check in about the findings I submitted—no rush at all, I completely understand you’ve had a lot going on. If there’s anything I can do to help with the review or make it easier, just let me know.

Thanks again for all the work you do on this project, and take care!

kyle-github, 2024-12-10:
Sorry, holidays and an unending set of migraines.   Such is life now.

zenozom, 2025-01-14:
Hi @kyle-github,

I hope you had a wonderful Christmas!

Do you have any update regarding the evaluation of the recurring pattern error?
If you still don't any update, we'd like to proceed with our disclosure process as reported [here](https://www.nozominetworks.com/legal/vulnerability-disclosure-policy).

Specifically, we'd like to complete the `Phase 2: Vendor Engagement` step by achieving the following tasks:
- Acknowledging/agreeing on the technical details about reported vulnerabilities (e.g., CVSS score and severity)
- Provide us the list of `libplctag` library versions that are affected by our findings
- Provide us a timeline regarding the release of the fix for these issues
- (**Optional**) Commercial or open-source products which internally use `libplctag`

kyle-github, 2025-01-16:
Ugh.  Yeah, this is on the top of my fix priorities.   I hope to have a release out by the end of January.  I think that the technical details you found are correct.  I might even raise the CVSS score a bit.   I went through the CVSS calculator and came up with 3.5.

I have found that the place you found where checks are not done is the worst and most obvious case.  I have not been able to craft an attack at lower levels.   There are several interacting checks such as only reading in a quantity of data equal to or less than the buffer size and checking that the sent packet size in the header is the same as the amount of data received.

The two places you found the problem are due to copy-and-paste coding used to set up a separate code path for a specific kind of PLC that is going to need changes that diverge from the main Rockwell-based code.  I went off on a tangent for the fix for this but I think I have something much simpler for it now.

I have not been able to go back all the way to check, but all versions that support the CIP 0x0A multiple packed request service are vulnerable.   I suspect all 2.x versions.

kyle-github, 2025-01-26:
End of January might not be feasible.   I am taking care of my mother 24/7 right now.   If I don't make January 31st.  It will be the week after.  There are a few other tweaks I want to make because they might possibly used as ways to attack the library.  I can't see how to exploit them, but they do cause errors.

kyle-github, 2025-03-04:
Late, late, late.  

I added more sanity checks that I think will fix this problem but I am still concerned that there is far too much wild pointer handling and I should move to a system using slices as it used in ab_server.    

Please reopen if this is still an issue with release 2.6.4 Beta!

closed 2025-03-04 by kyle-github