libplctag/libplctag #495: UDT read/write with string member(s)

dapittman, 2024-10-18:
I am using libplctag with a custom c# wrapper that correctly handles most data types, but I am struggling with UDTs that have string members.

See example usage below:

```csharp

// Create controller. Specify network
// address and slot number.
Controller controller = new("xxx.xxx.xxx.xxx", 0);

// Define tag with name "UDT_INSTANCE" as byte array size 52.
Tag<byte[]> udt = new(controller, "UDT_INSTANCE", 52);

// Read tag bytes.
byte[] bytes = udt.Value;

// Load bytes into instance.
Thing thing = new(bytes);

// Modify instance members. The appropriate
// slice of the underlying byte array is
// manipulated for each member
// by the property setters.
thing.DINT_MEMBER = 12345;
thing.STRING_MEMBER = "ABCDEFG";
thing.REAL_MEMBER = 123.456;

// Write tag bytes. There is
// an implicit operator to
// convert from Thing
// to byte[].
udt.Value = thing;

```

This correctly updates the bytes for the tag in the PLC. Unfortunately, the data in STRING_MEMBER will appear incorrect (depending entirely on whether or not the size of the string changed). In my example we wrote "ABCDEFG" into STRING_MEMBER, but if the string's DATA array previously contained "ABCDEFGHIJ" then we will see "ABCDEFG$00$00$00" since the LEN was not updated from 10 to 7. Alternatively, if the DATA array previously contained "ABCD" then we will still only see "ABCD" since LEN will be 4 (even though the rest of the bytes are there).

If I only need to read/write the string (not the entire UDT) I can handle it as follows:

```csharp

// Create controller. Specify network
// address and slot number.
Controller controller = new("xxx.xxx.xxx.xxx", 0);

// Define tag with name "UDT_INSTANCE.STRING_MEMBER" as string size 12.
Tag<string> stringMember = new(controller, "UDT_INSTANCE.STRING_MEMBER", 12);

// Read tag.
string value = stringMember.Value;

// Write tag.
stringMember.Value = "ABCDEFG";

```

In the second example the LEN always updates correctly and the internal string is fine. I guess my question is, what mechanism(s) does libplctag provide for handling UDTs?

Thanks

timyhac, 2024-10-18:
In Rockwell controllers, STRING types have two fields, the LEN field and the DATA field - both need to be updated whenever you update the value.

I'm not sure how your wrapper works but it could be something like this??

```csharp
Tag<string> stringMemberData = new(controller, "UDT_INSTANCE.STRING_MEMBER.DATA", 12);
Tag<uint> stringMemberLength = new(controller, "UDT_INSTANCE.STRING_MEMBER.LEN", 1);

stringMemberData.Value = "ABCDEFG";
stringMemberLength.Value = 7;
```

libplctag does nothing special here. It is the Rockwell Controller that is interpreting the tag name, and what the `.DATA` and `.LEN` portion of the Tag name mean, and what to do with the bytes you send to it.

---

Tag values are just byte arrays, so another way to to would be to operate on _both _ the LEN and DATA fields before sending to the PLC.

First, document the binary structure of your UDT so that you know which bytes do what.
In the case of STRING in Logix controllers:
* Bytes 0-3 are used to store the LEN field as uint32LE.
* Bytes 4-85 are used to store the ASCII-encoded string.
* Bytes 86-87 are padding.

```csharp
// .LEN = 7
plc_tag_set_uint8(MY_TAG_ID, 0, 0x07);
plc_tag_set_uint8(MY_TAG_ID, 1, 0x00);
plc_tag_set_uint8(MY_TAG_ID, 2, 0x00);
plc_tag_set_uint8(MY_TAG_ID, 3, 0x00);

// .DATA = "ABCDEFG"
plc_tag_set_uint8(MY_TAG_ID, 4, 0x41);   // "A"
plc_tag_set_uint8(MY_TAG_ID, 5, 0x42);   // "B"
plc_tag_set_uint8(MY_TAG_ID, 6, 0x43);   // "C"
plc_tag_set_uint8(MY_TAG_ID, 7, 0x44);   // "D"
plc_tag_set_uint8(MY_TAG_ID, 8, 0x45);   // "E"
plc_tag_set_uint8(MY_TAG_ID, 9, 0x46);   // "F"
plc_tag_set_uint8(MY_TAG_ID, 10, 0x47);   // "G"
```

---

Another way could be to use the [Tag Data Accessors](https://github.com/libplctag/libplctag/wiki/API#tag-data-accessors) to encode the appropriate values into it.

```csharp
byte[] myStr = [0x41, 0x42, 0x43, 0x44, 0x45, 0x46, 0x47]; // "ABCDEFG"

plc_tag_set_int32(MY_TAG_ID, 0, myStr.Length);
plc_tag_set_raw_bytes(MY_TAG_ID, 4, myStr, myStr.Length);
```

---

Lastly, there are specific [APIs for string handling](https://github.com/libplctag/libplctag/wiki/Strings-and-String-Handling).

I see you're developing your own wrapper but studying the examples from libplctag.NET should still help:

https://github.com/libplctag/libplctag.NET/blob/main/examples/CSharp%20DotNetCore/PlcMapper/ExampleCustomUdt.cs

https://github.com/libplctag/libplctag.NET/blob/main/examples/CSharp%20DotNetCore/PlcMapper/ExampleMappers.cs#L215

dapittman, 2024-10-18:
You have definitely given me something to think about. In hindsight it seems obvious that the LEN would be there in the struct. When I examined the full byte array I just thought there was excessive padding. The custom string in the UDT I was testing has a limit of 12 characters but I believe it was 16 bytes between the two neighboring fields. I will take a closer look at what happens to the bytes before and after I manually change the LEN in the PLC.

Thanks

dapittman, 2024-10-21:
@timyhac had the solution. This isn't a problem with libplctag.

I verified that the string LEN could be found in the 4 bytes directly ahead of the string DATA.

closed 2024-10-21 by dapittman