libplctag/libplctag #497: Segfault caused by double free for logixpccc

EchoHowardLam-GTS, 2024-11-11:
Hi,

Segfault can be consistently reproduced in the following condition:
1. cpu is set to logixpccc
2. there are more than one tag handle accessing the same virtual file at the same time
3. no debugging tool is enabled, compiled in release mode

Here is a code sample that 100% reproduce the segfault on my side:
```
#include <stdio.h>
#include <stdlib.h>
#include <stdarg.h>
#include "../lib/libplctag.h"
#include "utils.h"

#define REQUIRED_VERSION 2,4,0

#define STR_IMPL_(x) #x      //stringify argument
#define STR(x) STR_IMPL_(x)  //indirection to expand argument macros

#define TABLE_SIZE 50
#define TABLE_SIZE2 1
#define DATA_TIMEOUT 5000
#define TRY_TIMES 20
#define TIME_BETW_OP 100
#define TIME_BETW_TAG 0
#define PLC5000_PATH1 "protocol=ab-eip&elem_count=" STR(TABLE_SIZE) "&gateway=138.13.33.111&path=1,0&cpu=logixpccc&name=N100:0"
#define PLC5000_PATH2 "protocol=ab-eip&elem_count=" STR(TABLE_SIZE) "&gateway=138.13.33.111&path=1,0&cpu=logixpccc&name=N100:10"

static void FATAL(const char *format, ...) {
    va_list args;
    va_start(args, format);
    vprintf(format, args);
    va_end(args);
    exit(1);
}

void ConcurrentRead(const char *tagPath1, const char *tagPath2)
{
    int32_t tagHandles[2] = {
        plc_tag_create(tagPath1, DATA_TIMEOUT),
        plc_tag_create(tagPath2, DATA_TIMEOUT)
    };

    for (int i = 0; i < 2; i++)
        if (tagHandles[i] < 0)
            FATAL("Error[%d] creating tag handle: %s\n", i, plc_tag_decode_error(tagHandles[i]));

    for (int t = 0; t < TRY_TIMES; t++) {

        int rc[2] = { 0, 0 };
        int32_t elem_size[2] = { 0, 0 };

        // Invoke async read for both tag handles
        for (int i = 0; i < 2; i++) {
            rc[i] = plc_tag_read(tagHandles[i], 0);
            util_sleep_ms(TIME_BETW_TAG);
        }

        // Wait for both tag handles to be finish reading
        for (int i = 0; i < 2; i++) {
            while (rc[i] != PLCTAG_STATUS_OK) {
                if (rc[i] != PLCTAG_STATUS_OK && rc[i] != PLCTAG_STATUS_PENDING)
                    FATAL("Error[%d] %d: %s\n", i, rc[i], plc_tag_decode_error(rc[i]));

                util_sleep_ms(10);
                rc[i] = plc_tag_status(tagHandles[i]);
            }
        }

        // Print their values
        for (int i = 0; i < 2; i++) {
            elem_size[i] = plc_tag_get_int_attribute(tagHandles[i], "elem_size", 0);
            for (int j = 0; j < TABLE_SIZE2; j++)
                printf("Tag[%d][%d] = %d\n", i, j, plc_tag_get_int16(tagHandles[i], (j * elem_size[i])));
        }

        util_sleep_ms(TIME_BETW_OP);
    }

    for (int i = 0; i < 2; i++)
        plc_tag_destroy(tagHandles[i]);
    return;
}

int main()
{
    /* check the library version. */
    if(plc_tag_check_lib_version(REQUIRED_VERSION) != PLCTAG_STATUS_OK) {
        printf("Required compatible library version %d.%d.%d not available!", REQUIRED_VERSION);
        exit(1);
    }

    printf("Test: Sequentially Read Same Table\n");
    // SequentialRead(PLC5000_PATH1, PLC5000_PATH2);
    ConcurrentRead(PLC5000_PATH1, PLC5000_PATH2);

    printf("Test: finished successfully\n");
    return 0;
}
```

The following result can be observed on v2.5, release and prerelease.
```
~/work/libplctag-latest/build/bin_dist
Test: Sequentially Read Same Table
Tag[0][0] = 31
Tag[1][0] = 0
Tag[0][0] = 31
Tag[1][0] = 0
Tag[0][0] = 31
Tag[1][0] = 0
Tag[0][0] = 31
Tag[1][0] = 0
Segmentation fault
```

EchoHowardLam-GTS, 2024-11-11:
After investigation, the segfault is thrown by a malloc call because of memory corruption affecting the heap pointer.

By some comment-out tricks I can confirm the crash is caused by the exact same issue shown below.
**The memory corruption observed in a simpler code sample:**
```
#include <stdio.h>
#include <stdlib.h>
#include <stdarg.h>
#include "../lib/libplctag.h"
#include "utils.h"

#define REQUIRED_VERSION 2,4,0

#define STR_IMPL_(x) #x      //stringify argument
#define STR(x) STR_IMPL_(x)  //indirection to expand argument macros

#define TABLE_SIZE 50
#define TABLE_SIZE2 1
#define DATA_TIMEOUT 5000
#define TRY_TIMES 20
#define TIME_BETW_OP 100
#define TIME_BETW_TAG 0
#define PLC5000_PATH1 "protocol=ab-eip&elem_count=" STR(TABLE_SIZE) "&gateway=138.13.33.111&path=1,0&cpu=logixpccc&name=N100:0"
#define PLC5000_PATH2 "protocol=ab-eip&elem_count=" STR(TABLE_SIZE) "&gateway=138.13.33.111&path=1,0&cpu=logixpccc&name=N100:10"

static void FATAL(const char *format, ...) {
    va_list args;
    va_start(args, format);
    vprintf(format, args);
    va_end(args);
    exit(1);
}

void SequentialRead(const char *tagPath1, const char *tagPath2)
{
    int32_t tagHandles[2] = {
        plc_tag_create(tagPath1, DATA_TIMEOUT),
        plc_tag_create(tagPath2, DATA_TIMEOUT)
    };

    if (tagHandles[0] < 0) FATAL("Error creating tag handle 0: %s\n", plc_tag_decode_error(tagHandles[0]));
    if (tagHandles[1] < 0) FATAL("Error creating tag handle 1: %s\n", plc_tag_decode_error(tagHandles[1]));

    for (int t = 0; t < TRY_TIMES; t++) {
        for (int i = 0; i < 2; i++) {
            int rc = plc_tag_read(tagHandles[i], DATA_TIMEOUT);

            if (rc != PLCTAG_STATUS_OK)
                FATAL("Error tag handle %d: %d, %s\n", i, rc, plc_tag_decode_error(rc));

            int32_t elem_size = plc_tag_get_int_attribute(tagHandles[i], "elem_size", 0);
            for (int j = 0; j < TABLE_SIZE2; j++)
                printf("Tag[%d][%d] = %d\n", i, j, plc_tag_get_int16(tagHandles[i], (j * elem_size)));

            util_sleep_ms(TIME_BETW_TAG);
        }
        util_sleep_ms(TIME_BETW_OP);
    }

    plc_tag_destroy(tagHandles[0]);
    plc_tag_destroy(tagHandles[1]);
}

int main()
{
    /* check the library version. */
    if(plc_tag_check_lib_version(REQUIRED_VERSION) != PLCTAG_STATUS_OK) {
        printf("Required compatible library version %d.%d.%d not available!", REQUIRED_VERSION);
        exit(1);
    }
[... 76 lines cut ...]
==482185==    at 0x488ACEB: rc_dec_impl (in /home/echo/work/libplctag-latest/build/bin_dist/libplctag.so.2.6)
==482185==    by 0x86: ???
==482185==    by 0x486717B: tag_tickler (in /home/echo/work/libplctag-latest/build/bin_dist/libplctag.so.2.6)
==482185==    by 0x4850F1E: tag_tickler_func (in /home/echo/work/libplctag-latest/build/bin_dist/libplctag.so.2.6)
==482185==    by 0x4AA4EA6: start_thread (pthread_create.c:477)
==482185==    by 0x49C4ACE: clone (clone.S:95)
==482185==  Address 0x4ac3dc0 is 0 bytes inside a block of size 88 free'd
==482185==    at 0x48399AB: free (vg_replace_malloc.c:538)
==482185==    by 0x488ADAA: rc_dec_impl (in /home/echo/work/libplctag-latest/build/bin_dist/libplctag.so.2.6)
==482185==    by 0x4867160: tag_tickler (in /home/echo/work/libplctag-latest/build/bin_dist/libplctag.so.2.6)
==482185==    by 0x4850F1E: tag_tickler_func (in /home/echo/work/libplctag-latest/build/bin_dist/libplctag.so.2.6)
==482185==    by 0x4AA4EA6: start_thread (pthread_create.c:477)
==482185==    by 0x49C4ACE: clone (clone.S:95)
==482185==  Block was alloc'd at
==482185==    at 0x483AB65: calloc (vg_replace_malloc.c:760)
==482185==    by 0x488A9A6: rc_alloc_impl (in /home/echo/work/libplctag-latest/build/bin_dist/libplctag.so.2.6)
==482185==    by 0x487414A: session_create_request (in /home/echo/work/libplctag-latest/build/bin_dist/libplctag.so.2.6)
==482185==    by 0x4866167: tag_read_start (in /home/echo/work/libplctag-latest/build/bin_dist/libplctag.so.2.6)
==482185==    by 0x485CC7D: ab_tag_create (in /home/echo/work/libplctag-latest/build/bin_dist/libplctag.so.2.6)
==482185==    by 0x4851560: plc_tag_create_ex (in /home/echo/work/libplctag-latest/build/bin_dist/libplctag.so.2.6)
==482185==    by 0x10934A: SequentialRead (in /home/echo/work/libplctag-latest/build/bin_dist/pccc_minimal_crash)
==482185==    by 0x109157: main (in /home/echo/work/libplctag-latest/build/bin_dist/pccc_minimal_crash)
==482185==
...
```

EchoHowardLam-GTS, 2024-11-11:
The valgrind report suggests the program attempted to access to a freed memory

The following code is both the free site and invalid access site:

https://github.com/libplctag/libplctag/blob/c55bc5876d938dda1c609750cde5ae4812d7b8a8/src/protocols/ab/eip_lgx_pccc.c#L477-L490

The function assumes it is starting with request with at least 1 ref count,
then invoke rc_inc, rc_dec, rc_dec as the function goes and thereby end up freeing the request count it is holding.

For some reasons, the first rc_dec actually free the memory because its internal counter goes to 0.
Then the second rc_dec is using an invalidated data to free something again.

I suspect either the function starts with an rc that has counter=0,
or there is another thread calling too much rc_dec when the function is running. 

btw, I recall seeing the invalid access warning in valgrind report for other PLC types(i.e. controllogix),
but they didn't crash so I never prioritize checking them.

kyle-github, 2024-11-11:
Thanks for the great debugging!   This definitely looks like a double free.   That said, I think the error may be a little different than it appears.   If the count was 1 when the function is entered, then the rc_inc() should succeed and make the ref count 2 which should continue to work.   So I will need to trace the ref count from the initial creation to the end failure.

EchoHowardLam-GTS, 2025-02-07:
I got some more time to look into the issue and found something interesting.

With a slightly modified code to always print the func
Here is the log for logixpccc (where I encounter double free bug)
```
tag(0) WARN rc_inc_impl:185 ++ Ref count becomes 2 for 0x563093d0a730 at session_add_request_unsafe
tag(11) WARN rc_inc_impl:185 ++ Ref count becomes 3 for 0x563093d0a730 at check_read_status
tag(11) WARN rc_dec_impl:308 -- Ref count becomes 2 for 0x563093d0a730  at check_read_status
tag(0) WARN rc_dec_impl:308 -- Ref count becomes 1 for 0x563093d0a730  at process_requests
tag(0) WARN rc_inc_impl:185 ++ Ref count becomes 2 for 0x563093d0a730 at check_read_status
tag(0) WARN rc_dec_impl:308 -- Ref count becomes 1 for 0x563093d0a730  at ab_tag_abort
tag(0) WARN rc_dec_impl:308 -- Ref count becomes 0 for 0x563093d0a730  at check_read_status
tag(0) WARN rc_dec_impl:317 Calling cleanup functions due to call at check_read_status for 0x563093d0a730.
tag(0) WARN rc_dec_impl:308 -- Ref count becomes 0 for 0x563093d0a730 !!! at check_read_status
0x5630861e6848: (rc_dec_impl+0xd8)
```
The !!! is where double free happens.

In comparison, here is the log for controllogix (no double free bug)
```
tag(0) WARN rc_inc_impl:185 ++ Ref count becomes 2 for 0x55bed97bf6d0 at session_add_request_unsafe
tag(11) WARN rc_inc_impl:185 ++ Ref count becomes 3 for 0x55bed97bf6d0 at check_read_status_connected
tag(11) WARN rc_dec_impl:308 -- Ref count becomes 2 for 0x55bed97bf6d0  at check_read_status_connected
tag(0) WARN rc_dec_impl:308 -- Ref count becomes 1 for 0x55bed97bf6d0  at process_requests
tag(11) WARN rc_inc_impl:185 ++ Ref count becomes 2 for 0x55bed97bf6d0 at check_read_status_connected
tag(11) WARN rc_dec_impl:308 -- Ref count becomes 1 for 0x55bed97bf6d0  at check_read_status_connected
tag(11) WARN rc_dec_impl:308 -- Ref count becomes 0 for 0x55bed97bf6d0  at check_read_status_connected
tag(11) WARN rc_dec_impl:317 Calling cleanup functions due to call at check_read_status_connected:1498 for 0x55bed97bf6d0.
```

It can be seen that ab_tag_abort(...) is why logixpccc got an extra rc_dec and over-decremented itself
https://github.com/libplctag/libplctag/blob/c55bc5876d938dda1c609750cde5ae4812d7b8a8/src/protocols/ab/eip_lgx_pccc.c#L471-L490

EchoHowardLam-GTS, 2025-02-07:
This is what it would look like if eip_cip.c logic is used for logixpccc

https://github.com/EchoHowardLam-GTS/libplctag/commit/486ba89fb94b67a4c75e58c2bf43d3740b7e9730

tbf I am not sure why there is a ab_tag_abort needed at the end of function for both logixpccc and eip_cip (controllogix)
It seems that if that code is invoked, rc gets decremented three times in the function and could end up with double free.

Perhaps the function should avoid calling extra rc_dec if it decides to call ab_tag_abort?

kyle-github, 2025-03-04:
This needs more research.  There were a few small changes in 2.6.4 Beta.  Could you test that and see if it still reproduces the problem?   This is all far too complicated!  If you can still repro the problem, please reopen this!

EchoHowardLam-GTS, 2025-03-04:
Did a quick test. Valgrind starts complaining about invalid read once I switch to release branch
(tested commit 5be4ac305d9ded8fb5f6b0cc311125f687dff819)

```
==1535430== Invalid read of size 4
==1535430==    at 0xC6CF6C7: lock_acquire (in /home/echo/transfer/vfep/vfep/tmp/libplctag/build/bin_dist/libplctag.so.2.6)
==1535430==    by 0xC6CD4A9: rc_dec_impl (in /home/echo/transfer/vfep/vfep/tmp/libplctag/build/bin_dist/libplctag.so.2.6)
==1535430==    by 0xC6A9240: tag_tickler (in /home/echo/transfer/vfep/vfep/tmp/libplctag/build/bin_dist/libplctag.so.2.6)
==1535430==    by 0xC695D40: plc_tag_status (in /home/echo/transfer/vfep/vfep/tmp/libplctag/build/bin_dist/libplctag.so.2.6)
==1535430==    by 0xC6480D7: TagHandleManager::RetrieveTags(std::map<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int>, TagHandleManager::TagHandleSavedInfo, std::less<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int> >, std::allocator<std::pair<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int> const, TagHandleManager::TagHandleSavedInfo> > >&, std::set<int, std::less<int>, std::allocator<int> >*, std::vector<std::tuple<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int, int>, std::allocator<std::tuple<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int, int> > > const&, long) (ProtocolABCIP.cpp:1634)
==1535430==    by 0xC648F71: TagHandleManager::RetrieveReadTags(std::set<int, std::less<int>, std::allocator<int> >*, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, DatasetInfo, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, DatasetInfo> > > const&, ABCIPTagNamingStyle, long) (ProtocolABCIP.cpp:1533)
==1535430==    by 0x4866EA6: start_thread (pthread_create.c:477)
==1535430==    by 0x5485ACE: clone (clone.S:95)
==1535430==  Address 0x1118d640 is 0 bytes inside a block of size 88 free'd
==1535430==    at 0x48399AB: free (vg_replace_malloc.c:538)
==1535430==    by 0xC6CD5AA: rc_dec_impl (in /home/echo/transfer/vfep/vfep/tmp/libplctag/build/bin_dist/libplctag.so.2.6)
==1535430==    by 0xC6A9225: tag_tickler (in /home/echo/transfer/vfep/vfep/tmp/libplctag/build/bin_dist/libplctag.so.2.6)
==1535430==    by 0xC695D40: plc_tag_status (in /home/echo/transfer/vfep/vfep/tmp/libplctag/build/bin_dist/libplctag.so.2.6)
==1535430==    by 0xC6480D7: TagHandleManager::RetrieveTags(std::map<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int>, TagHandleManager::TagHandleSavedInfo, std::less<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int> >, std::allocator<std::pair<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int> const, TagHandleManager::TagHandleSavedInfo> > >&, std::set<int, std::less<int>, std::allocator<int> >*, std::vector<std::tuple<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int, int>, std::allocator<std::tuple<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int, int> > > const&, long) (ProtocolABCIP.cpp:1634)
==1535430==    by 0xC648F71: TagHandleManager::RetrieveReadTags(std::set<int, std::less<int>, std::allocator<int> >*, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, DatasetInfo, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, DatasetInfo> > > const&, ABCIPTagNamingStyle, long) (ProtocolABCIP.cpp:1533)
==1535430==    by 0x4866EA6: start_thread (pthread_create.c:477)
==1535430==    by 0x5485ACE: clone (clone.S:95)
==1535430==  Block was alloc'd at
==1535430==    at 0x483AB65: calloc (vg_replace_malloc.c:760)
==1535430==    by 0xC6CD1A6: rc_alloc_impl (in /home/echo/transfer/vfep/vfep/tmp/libplctag/build/bin_dist/libplctag.so.2.6)
==1535430==    by 0xC6B654A: session_create_request (in /home/echo/transfer/vfep/vfep/tmp/libplctag/build/bin_dist/libplctag.so.2.6)
==1535430==    by 0xC6A81D7: tag_read_start (in /home/echo/transfer/vfep/vfep/tmp/libplctag/build/bin_dist/libplctag.so.2.6)
==1535430==    by 0xC69EC6D: ab_tag_create (in /home/echo/transfer/vfep/vfep/tmp/libplctag/build/bin_dist/libplctag.so.2.6)
==1535430==    by 0xC693560: plc_tag_create_ex (in /home/echo/transfer/vfep/vfep/tmp/libplctag/build/bin_dist/libplctag.so.2.6)
==1535430==    by 0xC6486BF: TagHandleManager::RetrieveTags(std::map<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int>, TagHandleManager::TagHandleSavedInfo, std::less<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int> >, std::allocator<std::pair<std::pair<std::__cxx11::basic_string< [... cut ...]
==1535430==    by 0xC648F71: TagHandleManager::RetrieveReadTags(std::set<int, std::less<int>, std::allocator<int> >*, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, DatasetInfo, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, DatasetInfo> > > const&, ABCIPTagNamingStyle, long) (ProtocolABCIP.cpp:1533)
==1535430==    by 0x4866EA6: start_thread (pthread_create.c:477)
```

kyle-github, 2025-03-04:
Ugh.   Thanks for the repro.  I'll work on this.   Definitely a bug!

kyle-github, 2025-03-04:
There is definitely a problem with calling abort at this point.   However, there is also a problem with calling API functions within the library itself.   See #502.

kyle-github, 2025-03-05:
Huh.  Why did I do this only here?   And why are just the read status checks different?   This looks like a poor attempt at sharing code that should not have been shared.   The write paths look OK.  I think all that needs to be done is commenting out that `ab_tag_abort(tag)` call.   I will put that into the current release as this is a crash bug.

kyle-github, 2025-03-05:
Hopefully I got it this time!  Please try release [2.6.4-r2](https://github.com/libplctag/libplctag/releases/tag/v2.6.4-r2).

EchoHowardLam-GTS, 2025-03-06:
Trying release [2.6.4-r2](https://github.com/libplctag/libplctag/releases/tag/v2.6.4-r2) (Commit 8b787a399008288aee41a28f5cb5fe6fb9d66439)

It fails to connect now. Wireshark data is looking normal.
I tried to directly run cli with debug level 2 and it have some messages
```
thread(1) tag(0) WARN rc_inc_impl:166 Invalid pointer passed from check_read_status:378!
thread(1) tag(0) WARN check_read_request_status:1109 Read in progress, but no request in flight!
thread(1) tag(0) WARN rc_dec_impl:217 Null reference passed from check_read_status:382!
... (repeats above)
```

Let me spend some time debugging it.

EchoHowardLam-GTS, 2025-03-06:
https://github.com/libplctag/libplctag/blob/8b787a399008288aee41a28f5cb5fe6fb9d66439/src/protocols/ab/eip_lgx_pccc.c#L365-L384

In the first time running `check_read_status()`, the read attempt almost always fail as PLC cannot instantly reply
This causes line 382 geting executed in first poll.

The new change to set `tag->req = rc_dec(request);` now causes `tag->req` to become NULL as rc_dec always return NULL

In second poll to `check_read_status()`,
line 378 gets a `rc_inc(NULL)` and results in error, which the error propagates as all return values are now NULL

EchoHowardLam-GTS, 2025-03-06:
If I revert the line 382
`tag->req = rc_dec(request);`
to
`rc_dec(request);`

Then it can connect successfully and valgrind will not complain anything

Of course there might be some reason for that tag->req to exist...

EchoHowardLam-GTS, 2025-03-06:
I try modifying rc_dec to behave like rc_inc

```
void *rc_dec_impl(const char *func, int line_num, void *data)
{
    int count = 0;
    int invalid = 0;
    refcount_p rc = NULL;
    void *result = NULL;

    pdebug(DEBUG_SPEW,"Starting, called from %s:%d for %p",func, line_num, data);

    if(!data) {
        pdebug(DEBUG_WARN,"Null reference passed from %s:%d!", func, line_num);
        return result;
    }

    /* get the refcount structure. */
    rc = ((refcount_p)data) - 1;

    /* do this sorta atomically */
    spin_block(&rc->lock) {
        if(rc->count > 0) {
            rc->count--;
            count = rc->count;
        } else {
            count = rc->count;
            invalid = 1;
        }
    }

    if(invalid) {
        pdebug(DEBUG_WARN,"Reference has invalid count %d!", count);
        result = NULL;
    } else {
        pdebug(DEBUG_SPEW,"Ref count is %d for %p.", count, data);

        /* clean up only if count is zero. */
        if(rc && count <= 0) {
            pdebug(DEBUG_DETAIL,"Calling cleanup functions due to call at %s:%d for %p.", func, line_num, data);

            refcount_cleanup(rc);
        }
        result = (count <= 0) ? NULL: data;
    }

    return result;
}
```

Nope got segfault, seems that there is a reason why rc_dec has to always return NULL now.
Anyway I am running out of time today so I will leave it here for now.

kyle-github, 2025-03-06:
Thank you so much for testing!!!  I had a PLC with PLC/5 translation set up but I lost access to it a couple of years ago.  So this is hard to test. 

I think I am going to remove the internal call to the shared read status check.  All the weird code around taking a separate reference to the request is part of allowing the code to call that routine.  But that routine releases the request reference if there is a problem.   I think that the reference is still in PENDING mode at first and thus fails to check against PLCTAG_STATUS_OK.  I will look at the code more.  

Again, thank you for testing!

kyle-github, 2025-03-25:
OK.  I can use your program with 100 iterations and not see any problems.   I hope to release this within a week.  Thanks for being so patient and many thanks for finding this bug and creating a good repro!

kyle-github, 2025-04-11:
I have run this multiple times with ASan on and with Valgrind and cannot cause the fault any more.  Please reopen if version 2.6.4 does not fix this for you.  Thanks again for raising this issue and providing such a nice repro example!

closed 2025-04-11 by kyle-github