libplctag/libplctag #498: ab_server: issue with multithreaded writes

wegel, 2024-11-12:
Hi,

   I seem to have stumbled on a bug in ab_server for multithreaded writes. Again, I know that ab_server is only an internal tool, but there's so few small and (relatively) simple PLC simulator, I think it's worth it trying to fix it.

   This test fails: [test_multithreaded_write.c](https://github.com/wegel/libplctag/blob/ab_server-multithreaded-write/src/examples/test_multithreaded_write.c) (see [the CI run here](https://github.com/wegel/libplctag/actions/runs/11789713562/job/32838931228#step:17:18))

   The one passes: [test_sequential_write.c](https://github.com/wegel/libplctag/blob/ab_server-multithreaded-write/src/examples/test_sequential_write.c)

   Haven't had a chance to look at the cause yet, if I can find some time, I'll post my findings here.

hraftery, 2025-01-09:
I _think_ the problem here is that `ab_server` is not multi-threaded?

I think `test_multithreaded_write` is your creation? The failure occurs when you call `plc_tag_create()`, which is long and complex, and I wasn't actually able to find where/why it returns `PLCTAG_ERR_UNSUPPORTED`. But given that you specify a [non-zero timeout](https://github.com/libplctag/libplctag/wiki/API#creating-a-tag-handle), it's likely it's performing a read before returning. _ab_server_ will only accept one active connection at a time, so multiple concurrent reads will be ignored.

I don't know why all 5 threads fail though, so I could well be wrong here.

On the multi-threading issue, I agree `ab_server` is very close to the PLC simulator we desire, so I made it [multi-threaded](https://github.com/libplctag/libplctag/pull/503). Maybe you could try your test on that fork, prior to its release?

kyle-github, 2025-03-04:
I believe the underlying problem was the single-threaded nature of ab_server.  It is now multi-threaded, so please tests 2.6.4 Beta!