libplctag/libplctag #499: ab_server only accepts one connection at a time, so can't be used for PLC simulation

hraftery, 2024-11-18:
As a new libplctag user I was excited to find `ab_server` in solid use as a CI test tool, and that it has been suggested as a starting point for PLC simulation.

I've created a wrapper that launches `ab_server` with a set of tags, and then calls `tag_rw2` with `--write` every now and then to "play back" a scenario from a file. It works brilliantly, and I can see in the logs and with manual `tag_rw2` read calls that the server represents the scenario state at each point in time. A virtual PLC, presenting pre-meditated data streams, running on the PC!

Unfortunately, as soon as I launch my libplctag-enabled application and have it connect to the simulated PLC, the next `tag_rw2` call to update the scenario data hangs.

After some investigation, I see that libplctag works by opening a connection, queueing up requests, handling a transaction, queueing up requests, handling a transaction, and so on. Only after [5 seconds](https://groups.google.com/g/libplctag/c/g0Zwr0Kxwfc/m/xr43A0ZnBwAJ) will it opt to close the connection between requests.

Fair enough. But alas, `ab_server` only [handles one connection](https://github.com/libplctag/libplctag/blob/release/src/tests/ab_server/src/tcp_server.c#L78) at a time. So my dreams of a PLC simulator seem shot.

The only workaround I can find is to slow my application's polling rate to slower than once ever 5 seconds. Any other suggestions? As others have said, I don't want to burden ab_server's role in life with the requirements of a simulation tool, but it seems so very close!

hraftery, 2024-11-23:
Good to hear this was your intention anyway. In that case, it seems amiss of me not to contribute!

Honestly my knowledge of multi-threaded sockets servers in C was forged some 20 years ago, but your codebase is easy to work with and this came together really quickly. I've attempted to follow the code conventions as best I could, and leave clear implementation notes in the PR to aid integration and maintenance. Essentially it's just your `accept()` loop, except that it spawns a thread each time, and the mutable data is mutex protected. The mutable data at that point, turns out to be really well contained.

I've been using this for Real Work for a few days and it's working a treat - I can run a long lived `ab_server` instance, connect via the application's `libplctag`, and also periodically connect via `tag_rw2` to update tags.

No stress if you prefer to go at it another way, but at least there's a proof in practice and a solution that gets me out of trouble in the meantime.

kyle-github, 2025-03-04:
It might be, should be multithreaded now.  I still need to write more tests around this as I did not do the multithreading code.

closed 2025-03-04 by kyle-github