libplctag/libplctag #501: PLCTAG_ERR_NOT_FOUND on a tag that should exist

wegel, 2024-11-20:
I'd greatly appreciate some help on diagnosing something that happens but shouldn't be possible ;)

I do something like this (pseudo-code):
```
bool write_tag(char* tag_name, int data_type, int count, void* data) {
    // Create tag
    int tag = plc_tag_create(tag_name, TIMEOUT);
    if (tag < 0) {
        handle_error();
        return false;
    }

    // Copy data to tag using set_value_to_tag

    // Register callback
    if (plc_tag_register_callback_ex(tag, tag_callback, callback_data) != 0) {
        plc_tag_destroy(tag);
        return false;
    }

    // Write to PLC
    if (plc_tag_write(tag, TIMEOUT) != 0) {
        // Callback will handle cleanup on error
        return false;
    }

    return true;
}
```

then in `tag_callback`, on `PLCTAG_EVENT_WRITE_COMPLETED` I do some processing in a thread from a thread pool (eg, I try to not block `tag_callback`, then call:
```
plc_tag_unregister_callback(tag_id);
plc_tag_destroy(tag_id);
```

The problem is that sometimes, not that often, I get a sequence like this (actual log):

```
tag(20) created for WRITE: protocol=ab-eip&gateway=10.0.20.14&plc=controllogix&path=1,0&elem_size=4&elem_count=1&name=Test_tag[1]&debug=2
tag(20) calling plc_tag_write: protocol=ab-eip&gateway=10.0.20.14&plc=controllogix&path=1,0&elem_size=4&elem_count=1&name=Test_tag[1]&debug=2
tag(20) PLCTAG_EVENT_WRITE_STARTED
tag(20) PLCTAG_EVENT_WRITE_COMPLETED
tag(21) created for WRITE: protocol=ab-eip&gateway=10.0.20.14&plc=controllogix&path=1,0&elem_size=4&elem_count=1&name=Test_tag[1]&debug=2
tag(21) calling plc_tag_write: protocol=ab-eip&gateway=10.0.20.14&plc=controllogix&path=1,0&elem_size=4&elem_count=1&name=Test_tag[1]&debug=2
tag(21) PLCTAG_EVENT_WRITE_STARTED
2024-11-20 12:40:49.962 thread(1) tag(12) WARN plc_tag_generic_tickler:375 Skipping 2 periods of 100ms.
tag(21) PLCTAG_EVENT_WRITE_COMPLETED
tag(22) created for WRITE: protocol=ab-eip&gateway=10.0.20.14&plc=controllogix&path=1,0&elem_size=4&elem_count=1&name=Test_tag[1]&debug=2
tag(22) calling plc_tag_write: protocol=ab-eip&gateway=10.0.20.14&plc=controllogix&path=1,0&elem_size=4&elem_count=1&name=Test_tag[1]&debug=2
tag(22) PLCTAG_EVENT_WRITE_STARTED
tag(22) PLCTAG_EVENT_WRITE_COMPLETED
tag(23) created for WRITE: protocol=ab-eip&gateway=10.0.20.14&plc=controllogix&path=1,0&elem_size=4&elem_count=1&name=Test_tag[1]&debug=2
tag(23) calling plc_tag_write: protocol=ab-eip&gateway=10.0.20.14&plc=controllogix&path=1,0&elem_size=4&elem_count=1&name=Test_tag[1]&debug=2
tag(23) PLCTAG_EVENT_WRITE_STARTED
2024-11-20 12:40:50.259 thread(1) tag(12) WARN plc_tag_generic_tickler:375 Skipping 2 periods of 100ms.
tag(23) PLCTAG_EVENT_WRITE_COMPLETED
tag(24) created for WRITE: protocol=ab-eip&gateway=10.0.20.14&plc=controllogix&path=1,0&elem_size=4&elem_count=1&name=Test_tag[1]&debug=2
tag(24) calling plc_tag_write: protocol=ab-eip&gateway=10.0.20.14&plc=controllogix&path=1,0&elem_size=4&elem_count=1&name=Test_tag[1]&debug=2
tag(24) PLCTAG_EVENT_WRITE_STARTED
2024-11-20 12:40:50.420 thread(1) tag(12) WARN plc_tag_generic_tickler:375 Skipping 2 periods of 100ms.
tag(24) PLCTAG_EVENT_WRITE_COMPLETED
2024-11-20 12:40:50.491 thread(2) tag(0) WARN lookup_tag:4439 Tag with ID 24 not found.
2024-11-20 12:40:50.491 thread(2) tag(0) WARN plc_tag_status:1785 Tag not found.
2024-11-20 12:40:50.491 thread(2) tag(0) WARN plc_tag_write:1941 Error PLCTAG_ERR_NOT_FOUND while trying to write tag!
2024-11-20 12:40:50.491 thread(2) tag(0) WARN lookup_tag:4439 Tag with ID 24 not found.
2024-11-20 12:40:50.491 thread(2) tag(0) WARN plc_tag_abort:1501 Tag not found.
Operation: Write, Error: PLC_ERR_NOT_FOUND, Source: TagWrite, Tag Path: protocol=ab-eip&gateway=10.0.20.14&plc=controllogix&path=1,0&elem_size=4&elem_count=1&name=Test_tag[1]&debug=2
```

As you can see, for tag 24, the tag gets created and then `plc_tag_write` called, but the call returns an error (that's the line `Operation: Write, Error: PLC_ERR_NOT_FOUND...` . `PLCTAG_EVENT_WRITE_STARTED` and `PLCTAG_EVENT_WRITE_COMPLETED` are still called and, AFAIK, work as they should.

What I can't figure out is why/how `plc_tag_write` fails here; the tag has just been successfully created (see pseudo-code above).

Any idea? Any info I can provide to help diagnose?
Thanks a lot!

wegel, 2024-11-20:
It's of course only circumstancial, eg it doesn't prove that my problem is fixed, but adding a 2ms delay just before calling `plc_tag_destroy` *seems* to prevent the issue for me currently...

kyle-github, 2024-11-20:
General note: creating a tag handle (ID) could be a very heavy operation.   The first time a PLC is referenced the library will set up a connection to it.  That can take many milliseconds.  If you have at least one tag handle open all the time, subsequent ones will be fast and relatively cheap.   That said, tag handles are designed to be long lived (hours, days, weeks) so avoid creating and destroying them often.

Note that tag destruction is also potentially very heavy.  If it is the last tag that references that PLC, the process to clean up the CIP connection and all the handshakes necessary for that will be done and that is network traffic.   Calling plc_tag_destroy() can take many milliseconds.

I will post a short outline of the whole library lifecycle on the [forum](https://groups.google.com/g/libplctag) or make a wiki page sometime later today.

This smells a bit like a race condition.   It could be reuse of a tag ID internally in the library though that seems to be unlikely.  There are at least four threads at play here:

- your main application thread.
- the library maintains a single "tag tickler" thread that calls all tags to update their state.
- each PLC TCP connection has its own thread.
- whatever thread pool thread is doing some processing.

Depending on timing, each of these four threads could the the one that calls the callback.   That makes this a bit challenging to untangle!

The first thing I note is that you are doing something in your callback that results in at least 200ms of time.   The rows here show this.

```
tag(21) PLCTAG_EVENT_WRITE_STARTED
2024-11-20 12:40:49.962 thread(1) tag(12) WARN plc_tag_generic_tickler:375 Skipping 2 periods of 100ms.
tag(21) PLCTAG_EVENT_WRITE_COMPLETED
```

The tag tickler runs at least every 100ms.  It runs more often if something happens that wakes it up (like an event to process).  The "WARN" above shows that the tickler was hung on a mutex for at least 200ms. So somewhere your callback is locking something that occasionally takes at least 200ms to unlock.   The callback is not necessarily called from the tickler thread.  It could be that the callback is calling an API function (all API functions are protected by an internal mutex in the internal tag data or access static data) and the tickler is trying to call another API function on that same tag.

Please run the same test with debug level 4 so that we get a better idea of what is happening.  This will generate a lot of output!

That should show whether there is a problem in the library where tag IDs get reused too often or if it is a race condition like I think.

wegel, 2024-11-20:
I can confirm that, AFAIK, the callbacks are done in <1ms. FYI, my current ping to the PLC is ~75ms, and it often jumps above 100ms.

wegel, 2024-11-20:
Lost access to the PLC for the night (will regain access tomorrow), but I managed to repro something a similar using the `ab_server` simulator along with a very fast auto-sync read of 10ms and running for ~30s (this is just part of the log, obviously):

[debug_level3.txt](https://github.com/user-attachments/files/17837460/debug_level3.txt)

Unfortunately a level 3 debug as that's what seems to be the default of abserver for `--debug`, if necessary I'll modify it to set `debug=4`. I'd also like to make a self-contained repro that I could post; there's a chance I could manage to do that tomorrow.

wegel, 2024-11-21:
So this is clearly not really 1:1 and I wrote it pretty quick & dirty, but this respects (kind of) the general idea of what my actual current use-case scenario is:

https://github.com/wegel/libplctag/blob/diagnose-race-tag-not-found/src/examples/test_read_write_race_condition.c

You can see the CI failure here: https://github.com/wegel/libplctag/actions/runs/11946096290

There might obviously be something deemed incorrect/unsafe regarding the library usage in there; please tell me if you spot something.

Thanks!

kyle-github, 2024-11-21:
That latency with your PLC is horrible!!!!   75-100ms?  Yikes.   That's a problem but probably not the cause of the missing tag error.

Thanks for the log.  Tracking tag ID 176 (that's the one that first seems to be missing.  We see it created:

```
[ctest] 2024-11-20 17:16:23.675 thread(4) tag(0) INFO ab_tag_create:176 Starting.
[ctest] 2024-11-20 17:16:23.675 thread(4) tag(0) INFO rc_alloc_impl:111 Starting, called from ab_tag_create:188
[ctest] 2024-11-20 17:16:23.675 thread(4) tag(0) INFO rc_alloc_impl:130 Done
[ctest] 2024-11-20 17:16:23.675 thread(4) tag(0) INFO plc_tag_generic_init_tag:470 Starting.
[ctest] 2024-11-20 17:16:23.675 thread(4) tag(0) INFO plc_tag_generic_init_tag:501 Done.
[ctest] 2024-11-20 17:16:23.675 thread(4) tag(0) INFO get_tag_data_type:635 Setting element size to 4.
[ctest] 2024-11-20 17:16:23.675 thread(4) tag(0) INFO tag_read_start:290 Starting
[ctest] 2024-11-20 17:16:23.675 thread(4) tag(0) INFO build_read_request_connected:401 Starting.
[ctest] 2024-11-20 17:16:23.675 thread(4) tag(0) INFO rc_alloc_impl:111 Starting, called from session_create_request:2785
[ctest] 2024-11-20 17:16:23.675 thread(4) tag(0) INFO rc_alloc_impl:130 Done
[ctest] 2024-11-20 17:16:23.675 thread(4) tag(0) INFO session_add_request:1269 Starting. sess=000001C383EEC580, req=000001C383F150F0
[ctest] 2024-11-20 17:16:23.675 thread(4) tag(0) INFO session_add_request:1277 Done.
[ctest] 2024-11-20 17:16:23.675 thread(4) tag(0) INFO build_read_request_connected:486 Done
[ctest] 2024-11-20 17:16:23.675 thread(4) tag(0) INFO tag_read_start:319 Done.
[ctest] 2024-11-20 17:16:23.675 thread(4) tag(0) INFO ab_tag_create:516 Done.
[ctest] 2024-11-20 17:16:23.675 thread(4) tag(0) INFO set_tag_byte_order:4050 Starting.
[ctest] 2024-11-20 17:16:23.675 thread(4) tag(0) INFO set_tag_byte_order:4377 Done.
[ctest] 2024-11-20 17:16:23.675 thread(2) tag(0) INFO process_requests:1758 1 requests to process.
[ctest] 2024-11-20 17:16:23.675 thread(4) tag(176) INFO plc_tag_create_ex:979 Returning mapped tag ID 176
```

So it is created in thread 4.  Note that there is a read triggered.   The library triggers a "pre-read" so that the tag gets the latest data.  This way, once plc_tag_create() succeeds, the tag has data locally so if you immediately write as soon as the tag is ready, you don't write zero to the tag in the PLC.

Thread 4 triggers a write (the callback has been registered).

```
[ctest] 2024-11-20 17:16:23.675 thread(4) tag(176) INFO plc_tag_write:1836 Starting.
[ctest] 2024-11-20 17:16:23.675 thread(4) tag(0) INFO tag_write_start:340 Starting
[ctest] 2024-11-20 17:16:23.675 thread(4) tag(0) INFO build_write_request_connected:984 Starting.
[ctest] 2024-11-20 17:16:23.675 thread(4) tag(0) INFO rc_alloc_impl:111 Starting, called from session_create_request:2785
[ctest] 2024-11-20 17:16:23.675 thread(4) tag(0) INFO rc_alloc_impl:130 Done
[ctest] 2024-11-20 17:16:23.675 thread(4) tag(0) INFO session_add_request:1269 Starting. sess=000001C383EEC580, req=000001C383F15210
[ctest] tag(176) callback PLCTAG_EVENT_WRITE_STARTED
```

We see that a thread (not sure which one) calls the callback.

The request is sent by thread 2 and the response comes in on thread 2.

```
[ctest] 2024-11-20 17:16:23.676 thread(2) tag(176) INFO unpack_response:1866 Starting.
[ctest] 2024-11-20 17:16:23.676 thread(2) tag(176) INFO unpack_response:1875 Got single response packet.  Copying 50 bytes unchanged.
[ctest] 2024-11-20 17:16:23.676 thread(2) tag(176) INFO unpack_response:1974 Unpacked packet:
[ctest] 2024-11-20 17:16:23.676 thread(2) tag(176) INFO unpack_response:1975 00000 70 00 1a 00 c5 7a 00 00 00 00 00 00 00 00 00 00
[ctest] 2024-11-20 17:16:23.676 thread(2) tag(176) INFO unpack_response:1975 00016 00 00 00 00 00 00 00 00 02 00 a1 00 04 00 3b 0f
[ctest] 2024-11-20 17:16:23.676 thread(2) tag(176) INFO unpack_response:1975 00032 00 00 b1 00 06 00 00 00 b1 00 1c 00 ed 01 cd 00
[ctest] 2024-11-20 17:16:23.676 thread(2) tag(176) INFO unpack_response:1975 00048 00 00
```

So thread 2 has tag ID 176 in its state.  Based on what it is doing, this is the thread for the PLC connection.  So far, thread 4 and thread 2 know about tag 176.  Thread 2 is not a problem as it maintains a ref count until it does not need the tag anymore for results.

What's this?

```
[ctest] tag(176) callback PLCTAG_EVENT_WRITE_COMPLETED
[ctest] 2024-11-20 17:16:23.676 thread(5) tag(176) INFO plc_tag_unregister_callback:1305 Starting.
[ctest] 2024-11-20 17:16:23.676 thread(5) tag(176) INFO plc_tag_unregister_callback:1324 Done.
[ctest] 2024-11-20 17:16:23.676 thread(5) tag(176) INFO plc_tag_destroy:1554 Starting.
[ctest] 2024-11-20 17:16:23.676 thread(5) tag(176) INFO plc_tag_destroy:1590 Done.
```

Thread 5 is destroying the tag.   What is thread 5?

Then thread 4 start to try to access tag 176.  That fails because thread 5 destroyed the tag.

```
[ctest] 2024-11-20 17:16:23.676 thread(4) tag(0) WARN lookup_tag:4439 Tag with ID 176 not found.
[ctest] 2024-11-20 17:16:23.676 thread(4) tag(0) WARN plc_tag_status:1785 Tag not found.
[ctest] 2024-11-20 17:16:23.676 thread(4) tag(0) WARN plc_tag_write:1941 Error PLCTAG_ERR_NOT_FOUND while trying to write tag!
[ctest] 2024-11-20 17:16:23.676 thread(4) tag(0) WARN lookup_tag:4439 Tag with ID 176 not found.
```

I searched the whole log and the lines from thread 5 above are it.   Maybe it is one of the threads from your thread pool?

It looks like you have at least two threads that hold onto the tag ID.   The tag is created in thread 4.  Then thread 5 does something with it and destroys the tag.  Then thread 4 tries to access the tag again.   If you are handing off the tag handle to a worker thread that will end up destroying the tag, then thread 4 needs to not access it again.

BTW, I am not sure if you noticed but plc_tag_create_ex() lets you set the callback function and user data pointer while creating the tag.

I hope this helps.

wegel, 2024-11-21:
In the log, `thread 4` would be the `write_tag` function I wrote in my original post, eg it’s just calling create, register_callback and write. It’s not doing anything with the tag after calling `plc_tag_write`.

So this is incorrect:
> Then thread 4 tries to access the tag again.

You can kinda see it in this line:
```
[ctest] 2024-11-20 17:16:23.676 thread(4) tag(0) WARN plc_tag_write:1941 Error PLCTAG_ERR_NOT_FOUND while trying to write tag!
```
 This log happens inside `plc_tag_write`; that’s still the original one, there’s currently ever only just a single write per tag handle in that code.

Did you get the chance to check my test/repro in my post above?

Thanks for your help!

kyle-github, 2024-11-22:
I am tracing through the library code and I found this in plc_tag_write():

```c
            rc = cond_wait(tag->tag_cond_wait, (int)timeout_left);
            if(rc != PLCTAG_STATUS_OK) {
                pdebug(DEBUG_WARN, "Error %s while waiting for tag write to complete!", plc_tag_decode_error(rc));
                plc_tag_abort(id);

                break;
            }

            /* get the tag status. */
            rc = plc_tag_status(id);
```

First it waits on the condition variable for the tag.   Then if that returns without an error, it tries to get the status of the tag.   BUT, if the callback is called before plc_tag_write() reads the status then the status read would use a stale handle.

In the tag tickler thread function, the following code checks for complete writes:

```c
                            if(tag->write_complete) {
                                tag->write_complete = 0;
                                tag->write_in_flight = 0;
                                tag->auto_sync_next_write = 0;

                                // tag->event_write_complete = 1;
                                tag_raise_event(tag, PLCTAG_EVENT_WRITE_COMPLETED, tag->status);

                                /* wake immediately */
                                plc_tag_tickler_wake();
                                cond_signal(tag->tag_cond_wait);
                            }
```

Notice the last part: We signal the condition variable.

In the log we see:

```
[ctest] 2024-11-20 17:16:23.676 thread(2) tag(176) INFO unpack_response:1866 Starting.
[ctest] 2024-11-20 17:16:23.676 thread(2) tag(176) INFO unpack_response:1875 Got single response packet.  Copying 50 bytes unchanged.
[ctest] 2024-11-20 17:16:23.676 thread(2) tag(176) INFO unpack_response:1974 Unpacked packet:
[ctest] 2024-11-20 17:16:23.676 thread(2) tag(176) INFO unpack_response:1975 00000 70 00 1a 00 c5 7a 00 00 00 00 00 00 00 00 00 00
[ctest] 2024-11-20 17:16:23.676 thread(2) tag(176) INFO unpack_response:1975 00016 00 00 00 00 00 00 00 00 02 00 a1 00 04 00 3b 0f
[ctest] 2024-11-20 17:16:23.676 thread(2) tag(176) INFO unpack_response:1975 00032 00 00 b1 00 06 00 00 00 b1 00 1c 00 ed 01 cd 00
[ctest] 2024-11-20 17:16:23.676 thread(2) tag(176) INFO unpack_response:1975 00048 00 00
[ctest] 2024-11-20 17:16:23.676 thread(3) tag(0) INFO refcount_cleanup:256 Starting
[ctest] 2024-11-20 17:16:23.676 thread(3) tag(0) INFO refcount_cleanup:268 Done.
[ctest] tag(176) callback PLCTAG_EVENT_WRITE_COMPLETED
[ctest] 2024-11-20 17:16:23.676 thread(5) tag(176) INFO plc_tag_unregister_callback:1305 Starting.
[ctest] 2024-11-20 17:16:23.676 thread(5) tag(176) INFO plc_tag_unregister_callback:1324 Done.
[ctest] 2024-11-20 17:16:23.676 thread(5) tag(176) INFO plc_tag_destroy:1554 Starting.
[ctest] 2024-11-20 17:16:23.676 thread(5) tag(176) INFO plc_tag_destroy:1590 Done.
[ctest] 2024-11-20 17:16:23.676 thread(4) tag(0) WARN lookup_tag:4439 Tag with ID 176 not found.
[ctest] 2024-11-20 17:16:23.676 thread(4) tag(0) WARN plc_tag_status:1785 Tag not found.
```

I think that thread 2 is the PLC connection.  It gets the result from the PLC and flags the request as completed.   That is probably what triggers the thread 3 (which I think is the tag tickler thread) to delete the request after setting the tag state to show the read complete and raise the write complete event and signals the tag condition variable.  The OS could switch threads at that point.  

This might be a bug!  My current theory is that something is signalling the condition variable and then plc_tag_write() is calling plc_tag_status() on a stale handle.   What is weird is that we should already have an extra ref count due to the entry into plc_tag_write().  That should ensure that it is safe.

Let me poke at this some more...   Events were retrofitted years after the main part of the library was built and they have some sharp edges.   If you get to it before I do, could you run your program with debug level 4?   That shows more detail about what functions are called.   The current level only shows some parts of the call chain.

kyle-github, 2024-11-22:
I am calling this a bug until proven otherwise.

kyle-github, 2024-11-22:
Ah, I think I found part of the problem.   plc_tag_destroy() removes the tag handle from the hash table.  So the next call to an API function that wants to look up the tag will fail.  I think it is _not_ the ref count that is the problem.  The behavior of plc_tag_destroy() is intentional.   

So if the thread in the thread pool calls plc_tag_destroy() at wall-clock time T and the main thread calls plc_tag_status() at time T+1, then the status call will fail because the ID is not known.

So I am back to my original thought that this is a race condition between your threads and the internal threads in the library.   That said, I need to go through the code carefully to make sure that I am not calling API functions internally and not checking the return code.

The sleep probably causes the OS to switch threads.   It at least delays enough that the original thread finishes up the plc_tag_write() call.

wegel, 2024-11-22:
I'm pretty sure I had the problem even when I executed `plc_tag_destroy` directly in the callback (eg, not in another thread). Will try to validate. But yeah your current hypothesis is pretty much exactly what I said in my OP:

> Currently it feels like plc_tag_destroy is called in my callback just before this line in plc_tag_write [...] here: https://github.com/libplctag/libplctag/blob/c55bc5876d938dda1c609750cde5ae4812d7b8a8/src/lib/lib.c#L1937

kyle-github, 2024-11-22:
Yep, you did get it right, but I think (still not convinced 100% either way) that this is not a bug in the library.   You have two threads that act on a tag thinking the tag handle is valid.   One thread destroys the tag handle then the other uses it.   That's a logic/race condition in your app's code.  That said, it would be nice if the library handled this better.   

I had ref counts driving the removal of tags from the lookup table but that proved to be a problem.  What could happen was that one thread would decrement the count and then try to remove the tag from the table.   But another thread could look up the tag just before the entry was removed.  That thread could get an invalid memory pointer.   Read after free :-(

I am going to be out for a while this morning but I will try to run your sample code.  I have a laptop on wifi and it has wildly varying ping times to a PLC so that might also simulate your long latency.

wegel, 2024-11-22:
> You have two threads that act on a tag thinking the tag handle is valid. One thread destroys the tag handle then the other uses it. That's a logic/race condition in your app's code.

Hmm, personally I wouldn't frame it like that, because 1) the library asks us to minimize the time we spend inside the callback, that's the sole reason I'm processing from within a thread  launched by the callback and 2) I did ask if it was valid to call `plc_tag_destroy` from the callback in other issue/question, and was answered that it is indeed safe.

And, again, there's no `... then the other uses it`. The "other" is the original thread, and it doesn't do anything after calling `plc_tag_write`.

To reiterate:
```
thread 1:
   plc_tag_create_tag_ex
   plc_tag_write

thread 2:
   plc_tag_destroy
```

That's it.
I can obviously fix the problem by putting a lock around `plc_tag_write` and `plc_tag_destroy`, but that feels bad.

kyle-github, 2024-11-22:
The fact that the error occurs within one of the main API functions is why I am on the fence about this being a library bug.   While the library cannot do everything to make app code correct, it should make it as easy as possible to do the right thing and not crash if something else is done.

I think I am going to call this a bug.   There is a relatively simple fix for this.  The code needs to get the status without looking up the ID.   We already have a pointer to the memory for the tag and we have an incremented ref count so nothing is going to free it out from underneath us.

```c
            /* get the tag status. */
            rc = plc_tag_status(id);
```

Needs to be something like:

```c
critical_block(tag->api_mutex) {
        /* get the tag status. */
        if(tag->vtable && tag->vtable->status) {
            rc = tag->vtable->status(tag);
        } else {
            /* need a better error here */
            rc = PLCTAG_ERR_NOT_IMPLEMENTED;
        }
}
```

That skips the look up and uses the existing pointer.

kyle-github, 2024-11-22:
Depends on fix for #502.

kyle-github, 2025-03-04:
I am going to close this because #502 should fix it.  This is a larger problem than you found!

closed 2025-03-04 by kyle-github