libplctag/libplctag #505: Issue reading tags of the predefined type Message

jggri, 2024-12-09:
Hi,

I am trying to read this Controller tag
[image]
The type MESSAGE is a predefined type with the following definition
[image]

If I use the tag name (in this case, ReadEthernetMSG), it doesn’t work. I get the error PLCTAG_ERR_NOT_ALLOWED, and when I analyze it with Wireshark, I see General status -> Privilege violation. However, if I access the tag using the field name, for example, ReadEthernetMSG.Flags, it works.

Is there a special format or configuration required to read this type? I am interested in reading all the field simultaneously

Thanks.

jggri, 2024-12-09:
External Access = Read/Write

kyle-github, 2024-12-09:
Thanks for using libplctag.   Can you please set debug level to 4 and capture the log so that I can see the bytes that are coming back from the PLC?   Something looks a bit fishy here.  The code might be translating the error incorrectly.   Normally an error of 0x08 (unsupported command) results in trying the old ForwardOpen, not in a privilege violation.

jggri, 2024-12-09:
Thanks kyle. Attached logs and wireshark 
[image]
[image]
[log_reading_field_ok.txt](https://github.com/user-attachments/files/18061500/log_reading_field_ok.txt)
[log_reading_tag.txt](https://github.com/user-attachments/files/18061501/log_reading_tag.txt)

kyle-github, 2024-12-09:
Thanks for the logs.   The PLC is definitely failing the CIP command in the second log file.

According to my translation table error code 0x0f means:

```
0x0F - Privilege violation - A permission/privilege check failed
```

The CIP response is this:

```
d2 00 0f 00
```

0xd2 is the byte that says that the read is done (0x80 | 0x52).   0x0f is the status byte.

So the PLC really is responding with an error that there is a privilege violation.   That is strange that the privs seem to be set for Read/Write and it is failing.   Is there some other layer of protection?

I see nothing wrong with the tag read sent to the PLC.  It all looks good.

jggri, 2024-12-09:
Yes, is very strange, because some predefined types work and others don't (same configuration and privileges).  I have checked with RSLinx Classic Lite -> Data Monitor, and it works. It might be using a different access method, or it could be accessing each field individually for every read (this approach works with the library using <tagName>.<tagFieldName>)
[image]

jggri, 2024-12-11:
I have tested all the predefined types available in Studio 5000, and I encounter the same issue with the following types:

    ALARM_ANALOG
    ALARM_DIGITAL
    AXIS_CIP_DRIVE
    COORDINATE_SYSTEM
    MESSAGE
    MOTION_GROUP

getting a privilege violation error when using the tag name directly

jggri, 2024-12-11:
The library is working fine

kyle-github, 2024-12-12:
I found a PLC with a program that has MESSAGE tags and I get exactly the same error.   It isn't your program.   I've dug into the response from the PLC and it is definitely a privilege error that is being returned.

kyle-github, 2024-12-12:
This is really a concern!   If you don't mind I would like to reopen this as a bug/enhancement to track what is learned.

kyle-github, 2024-12-12:
That is a possibility but I currently have the library set up to _not_ list out tag definitions automatically and it has no way to use them internally.   That said, I strongly suspect that the Rockwell tools must use some other way to read this data.

kyle-github, 2024-12-12:
The library uses the symbolic/name-based method to access tags.  There are ways to access tags based on the class and instance IDs of the tags.  Those are not supported by the library yet.  Sounds like they need to be!

kyle-github, 2025-03-04:
This depends on #200.

I am going to close this one for now as #200 already has most of the underlying requirements to fix this.

closed 2025-03-04 by kyle-github