libplctag/libplctag #506: Performance Issue: Limitation of 500 bytes per request

jggri, 2024-12-11:
Hi,

I'm trying to measure how many tags per second my application can read. During testing, I noticed a limitation in CIP layer to 500 bytes. 
[image]
Is there any configuration or setting that allows increasing this value?

jggri, 2024-12-17:
[image]
[image]

kyle-github, 2024-12-17:
It depends what kind of PLC and what level of firmware you have.   Newer Rockwell Control and CompactLogix will support the larger sizes (the ethernet module on a ControlLogix also impacts this).

The library tries to use the Forward Open Extended CIP command first.  If that fails, then it drops back to the old Forward Open.  The extended version supports asking for a packet size up to about 4k (8k for some Omron PLCs).   The old Forward Open can only negotiate up to about ~500 bytes.

Note that Micro800 series PLCs cannot negotiate the larger size IIRC.

jggri, 2024-12-17:
Thanks Kyle. That's what happening
[image]

closed 2025-01-16 by kyle-github