libplctag/libplctag #507: plc_tag_decode_error incorrect results

pjhrath, 2024-12-23:


kyle-github, 2024-12-23:
Can you show example code here?   The decode function is simply a big switch/case statement.   I do not understand how it could give different results.

pjhrath, 2024-12-24:
int CABPlc::WriteBitValue(int32_t id, CCriticalSection* cs, int16_t value)
{
	int				iRet;
	int				status;
	int				len;

	CString			strData;

	// ****** Testing No writes to all of the Plcs remove to go live

	if (m_bNoPlcWrites == true)
		return(PLCTAG_STATUS_OK);

	// End of Testing

	// Read the tag value 

	iRet = plc_tag_read(id, _TIMEOUT);
	if (iRet != PLCTAG_STATUS_OK) {
		strData.Format(_T("plc_tag_read Error Status %s %d - %s"), m_strDefns[id], iRet, plc_tag_decode_error(id));
		LOGSTRING(strData);
		return (plc_tag_status(id));
	}

	//	Update the value
	
	iRet = plc_tag_set_bit(id, 0, value);
	if (iRet != PLCTAG_STATUS_OK) {
		strData.Format(_T("plc_tag_set_bit Error Status %s %d - %s"), m_strDefns[id], iRet, plc_tag_decode_error(id));
		LOGSTRING(strData);
		return (plc_tag_status(id));
	}

	// A solicited write for the PLC

	status = cs->Lock();

	iRet = plc_tag_write(id, _TIMEOUT);
	cs->Unlock();
	if (iRet != PLCTAG_STATUS_OK)
	{
		strData.Format("PLC Tag Write bit Error Status %s %d - %s", m_strDefns[id], iRet, plc_tag_decode_error(id));
		LOGSTRING(strData);
		return (iRet);
	}

	return (iRet);
}

timyhac, 2025-01-01:
Looks like `plc_tag_decode_error` is being passed `id` instead of `iRet`.

[`plc_tag_decode_error`](https://github.com/libplctag/libplctag/wiki/API#printing-status-codes) expects the integer value of an error, not a tag handle.
[`plc_tag_create`](https://github.com/libplctag/libplctag/wiki/API#creating-a-tag-handle) returns an error code _instead of_ a tag handle if there is an error.
If you want to use the tag handle to get the status you can use [`plc_tag_status`](https://github.com/libplctag/libplctag/wiki/API#retrieving-tag-status).

kyle-github, 2025-01-01:
Ah!  Good catch.  I was gearing up to do some testing.  Hmmm...  Not sure how to fix this so that it can be caught.  There are some tricks I can use with IDs that might make some sense but I have too many other things on my plate right now.

kyle-github, 2025-01-01:
I am thinking about using a few bits in the i32 that I use for handles so that I can have a type field.   Even three bits would be fairly good.  I wish there was something more intelligent I could do with the bits though.

pjhrath, 2025-01-02:
Thanks for the catch. 

Kyle, I don't think that a change is necessary.  The error arose due to a misdone modification of a cut / paste of code from one routine that used the plc_tag_status routine to get the error detail where the return from the routine was a tag rather than a status to another situation where the error value was already available and just needed to be translated.  The documentation is clear for plc_tag_decode_error and really I should have noticed the mistake when doing the very first implementation of the code

Thanks a lot for the help.  Much appreciated.

kyle-github, 2025-01-02:
I think we can close this.   I still want to make it even harder to make this kind of mistake!

closed 2025-01-02 by kyle-github