libplctag/libplctag #509: Make available data type for ControlLogix tags

Hector-ER, 2024-12-31:
Hello,
   I was studing your code, because I want to detect the data type of the AB tag. I saw that when the tag is initialized you perform a Read, and then, in the answer you have some bytes that inform the data type and size.  You use this info to know the size of the buffer and also I can get the size, but I think that it's not available for me (user of the library) any information about data type. Am I right?
Is it possible to make it available?

Best regards and Happy New Year.

Héctor.

timyhac, 2025-01-01:
libplctag v2.6.0 exposed this with [`plc_tag_get_byte_array_attribute` and the `raw_tag_type_bytes` array attribute](https://github.com/libplctag/libplctag/wiki/API#tag-internal-attribute-functions).

I note that you're using the libplctag.NET wrapper - there is an example of using the above [here](https://github.com/libplctag/libplctag.NET/blob/main/examples/CSharp%20DotNetCore/ExampleTagDynamic.cs#L20) but a warning that I have not tested this.

Hector-ER, 2025-01-01:
Thanks, this is exactly whay I was looking for.

There is a very small bug (in libplctag.NET).  In method GetByteArrayAttribute native plc_tag_get_byte_array_attribute function is called, and if it's ok it returns with the length of the buffer.  In the following line ThrowIfStatusNotOk(result) is called which will throw an exception with the result.  "libplctag.LibPlcTagException: '2'"
Except for this (in my very quick test) it's working fine.

timyhac, 2025-01-01:
I didn't realize that but I see from [here](https://github.com/libplctag/libplctag/blob/c55bc5876d938dda1c609750cde5ae4812d7b8a8/src/protocols/ab/ab_common.c#L936) that you are correct.

I will copy the [logic from tag_rw2](https://github.com/libplctag/libplctag/blob/c55bc5876d938dda1c609750cde5ae4812d7b8a8/src/examples/tag_rw2.c#L914) and check the return value is <0 before assuming it is an error.

closed 2025-01-16 by kyle-github