libplctag/libplctag #510: ab_server mixes up tags that have the same prefix

hraftery, 2025-01-08:
Issue title is yet to be proven, but is the current theory, after noticing that writing to a tag named `WORD1_WORD2_WORD3_WORD4_ACK` seems to overwrite the value in tag named `WORD1_WORD2_WORD3_WORD4`.

The issue seems to be here:

https://github.com/libplctag/libplctag/blob/release/src/tests/ab_server/src/slice.h#L58

which is called here:

https://github.com/libplctag/libplctag/blob/release/src/tests/ab_server/src/cip.c#L643

To wit, tag names are matched by comparing the strings, but only up to the number of characters in the existing tag?

---

There are several other confounding factors that I hope prove irrelevant once I investigate the smoking gun above:

1. I'm writing from one client and reading from another (using my multi-threaded fork).
2. Quickly (read/writes occur every 50ms or so, asynchronously).
3. The `ACK` tag does not exist on the server - I'm expecting the write operation to just get dropped.

kyle-github, 2025-01-08:
Sigh.   I suspect you are correct.   It only checks the length of the shorter of the slice or string.   Sheesh.  Did I do anything right on that code.  I did write it originally in about three days during a vacation (my wife was not happy) and I think I am now having to make up for all the time I did not plan it out.

kyle-github, 2025-03-04:
This should be fixed in release 2.6.4.  Please reopen if you find errors!

closed 2025-03-04 by kyle-github