libplctag/libplctag #512: Duplicate Values and Data Loss with AutoSyncReadInterval in Allen-Bradley ControlLogix PLC Integration

kais123456789, 2025-01-08:
I am using this open-source library with its c# wrapper to read 300 tags from an Allen-Bradley ControlLogix PLC. The PLC updates the tags every 2 seconds, and I'm using the AutoSyncReadInterval to 700 ms, while decoding the data every 2 seconds. However, I am encountering an issue where duplicate values appear, likely due to data loss. For example, the expected incrementing values (1, 2, 3, 4, 5, 6, 7) are sometimes replaced with duplicates in the sequence, like 1, 2, 3, 3, 5, 6, 7. Has anyone else experienced a similar problem with this toolkit?

cyrine3110, 2025-01-08:
I have encountered the same issue where duplicated values appears instead of the accurate data which causes skipped values.

kyle-github, 2025-01-08:
The underlying protocol is TCP which will not allow for data loss at all, without completely dropping the connection.   There is a test run that uses multiple threads to read and write a single tag (the writers update the value) and there is no loss seen there (against real PLCs).  But that may only mean that the test is not testing the same scenario as this. 

Can you describe more about the platform, the version, the PLC type etc. that you are using?  Are you using the read caching setting? [read_cache_ms](https://github.com/libplctag/libplctag/wiki/All-Attributes#read_cache_ms)

If this is relatively easy to reproduce, can you reproduce it with debug level 4 set?   If you can, please capture the logs (there will be a lot of data!) and attach them to this issue.  If that is not possible, can you set up a callback that is called when the tag finishes reading and output the value and the millisecond count each time?   That log may show that there is a missing read or that there is some delay happening.

Do you have a lot of tags in use?   I have tested with tens of thousands of tags without a problem, but those tests do not test exactly this scenario either.

Thanks for bringing this up!

kais123456789, 2025-01-09:
We are reading 200 tags from a Controllogix PLC. The tags have been defined in the sample source code as follows:

var myTag = new NotifyValueChangedTag<DintPlcMapper, int>()
{
    Name = "D_Int1",
    Gateway = "192.168.2.33",
    Path = "1.0",
    PlcType = 0,
    Protocol = Protocol.ab_eip,
    AutoSyncReadInterval = 500,
    AutoSyncWriteInterval = 1000,
    Timeout = 1000,
    ReadCacheMillisecondDuration = 100  // read_cache_ms value
};

We are reading the tags as follows:

// Create the handle for each tag. Example for D_Int1
int handle = plc_tag_create("protocol=ab_eip&gateway=192.168.2.33&path=1,0&plc=controllogix&elem_size=1&name=D_Int1&auto_sync_read_ms=500&auto_sync_write_ms=1000&debug=4", 1000);

Then, we call this method to read the data for each tag:

plc_tag_get_int32(handle, 0);

We perform the above steps for all 200 tags.

Please note that we are not recreating the handles for each data read. However, when we call plc_tag_get_int32 every 2000 ms, we are receiving duplicate values. We have confirmed that the values exist on the RTU and we are able to view them using another Allen-Bradley client. When we encounter duplicate values, there seems to be a skipped value.

kais123456789, 2025-01-09:
I set the timeout to 0 on plc_tag_create and observed some improvements, but I am still experiencing issues with duplicate data and data loss. Are there other configuration parameters that can be fine-tuned to address this problem? Could the use-connected-message and allowPacking attributes help resolve it? If so, what would be the optimal combination of configuration settings?

Additionally, is it safe to call the plc_tag_get_int32 method in parallel with 100 threads? Is the method thread-safe, and would this approach cause any crashes or performance issues?

kyle-github, 2025-01-10:
I am getting a bit confused here.   

- The first code snippet creating the tags is using the .NET wrapper to create a .NET object.   The name `NotifyValueChangedTag` indicates that the application will be notified if the value of the underlying tag changes.   That is not part of the core library.  Could this be the layer where changes are missed?  How are you capturing the notification of changed value?
- I am not sure what you are trying to do in the second code snippet.   That creates a completely new handle to the same tag as the first code snippet, but uses the lowest level API.   You are not setting a callback so how do you know if the read has completed or not?
- I am not sure why you are setting the auto sync timing for writes.  That just delays a write that many milliseconds after the core library notices that the application changed the local copy of the tag value.   Are you also writing to this tag at the same time?

Can you describe more about the logic of your application?   The code snippets are confusing to me because you appear to be using both the .NET wrapper and the core C DLL at the same time.   

All API functions in the core C DLL are thread safe. You can call any of the data accessor functions from multiple threads.

The authors of the .NET wrapper have an [example](https://github.com/libplctag/libplctag.NET/blob/main/examples/CSharp%20NativeImport/CallbackEx.cs) of how to use a callback.  You can start with that and add the auto read tag attributes to trigger it automatically and work your way up to a full implementation.

kyle-github, 2025-04-11:
Please reopen this if it continues.   If you have samples that reproduce the problem with the core C DLL, please add them here!   I very much want to fix any bugs!

closed 2025-04-11 by kyle-github