libplctag/libplctag #513: Writing to an array of UDTs

OlClown, 2025-01-13:
Firstly, want to say, this is an awesome library, great job!

I am curious how one goes about writing to an array of UDTs.  The PLC Support section says "raw support" for writing UDTs but the data must be pulled out piece by piece.  I think I have an vauge idea of what that means but I did not see any examples of writing UDT arrays so I am hoping for some clarification.

Just for example, lets say I had a UDT with 3 values of different types like an int16, float, and string.  How might I go about writing to an array of those?  

Please let me know if I simply missed something in the example code provided or if more information is needed to answer my question.

Thanks.

kyle-github, 2025-01-13:
By raw support, I mean that you can read an array of UDTs or a single UDT as a single unit but you need to decode it by using the data accessor functions.   That means that you need to understand all the alignment and padding rules.   What I generally have done is to make a tag of the type of the UDT.   Then I zero out all the fields.   Then I read the whole thing.  Then I change one field at a time and look at the binary data to see where the field is.   You can use the tag_rw2 tool to do this by setting the output type to uint8.

Generally Rockwell alignment rules are that a field is aligned on the size of its type.  So an INT field is aligned on two-byte addresses.  A DINT is aligned on four byte addresses etc.   Things like strings get tricky.

BOOL fields in a UDT are dealt with by compressing them into single bits in a byte.   So a UDT that looks like this:

* flag1: BOOL
* counter1: INT
* flag2: BOOL

Would be laid out in memory like this:

|Byte|Content|
| - |- |
|0|bit 0 is flag1, bit 1 is flag2
|1|padding/used for alignment
|2|LSB of counter1
|3|MSB of counter1

For an array, you need to determine the size of the UDT and then loop over the whole set of tag data to get to each array element.   

I hope this makes sense.

Here's some (untested) code that would read an array of the UDT above:

```c
#define MAX_ELEMENTS (3)
#define UDT_SIZE_IN_BYTES (4)

struct {
     bool flag1;
     int16_t counter1;
     bool flag2;
} udt_array[MAX_ELEMENTS] = {0};

tag_handle = plc_tag_create("protocol=ab-eip&gateway=10.1.2.3&path=1,0&plc=ControlLogix&elem_count=3&name=myUDTArray", 5000);

/* read the whole array */
rc = plc_tag_read(tag_handle, 5000);

for(int element_index=0; element_index < MAX_ELEMENTS; element_index++) {
    int base_offset = element_index * UDT_SIZE_IN_BYTES;

    /* copy data into the C UDT array. */
    udt_array[element_index].flag1 = plc_tag_get_bit(tag_handle, base_offset * 8);
    udt_array[element_index].counter1 = plc_tag_get_int16(tag_handle, base_offset + 2);
    udt_array[element_index].flag2 = plc_tag_get_bit(tag_handle, (base_offset * 8) + 1);
}

```

To write out the array, you would use the data setters.   I'll assume that the code to change the values in the C array is obvious.

```c
for(int element_index=0; element_index < MAX_ELEMENTS; element_index++) {
    int base_offset = element_index * UDT_SIZE_IN_BYTES;

    /* copy data from the C UDT array into the tag data buffer. */
    plc_tag_set_bit(tag_handle, base_offset * 8, udt_array[element_index].flag1);
    plc_tag_set_int16(tag_handle, base_offset + 2, udt_array[element_index].counter1);
    plc_tag_set_bit(tag_handle, (base_offset * 8) + 1, udt_array[element_index].flag2);
}

/* write the whole array */
rc = plc_tag_write(tag_handle, 5000);
```

OlClown, 2025-01-14:
Thanks Kyle,

This answers my question!

kyle-github, 2025-01-15:
Can we close this?

closed 2025-01-16 by kyle-github