libplctag/libplctag #514: Error PLCTAG_ERR_TOO_LARGE Maximum Buffer Size

jggri, 2025-01-16:
I get 2025-01-16 18:08:09.820 thread(1) tag(0) WARN recv_eip_response:2270 Packet response (8746) is larger **than possible buffer size (4046)!** when trying to read this tag. 
2025-01-16 18:08:09.820 thread(1) tag(0) WARN process_requests:1748 Error receiving packet response PLCTAG_ERR_TOO_LARGE!
2025-01-16 18:08:09.820 thread(1) tag(0) WARN session_handler:1449 Error while processing requests PLCTAG_ERR_TOO_LARGE!
2025-01-16 18:08:09.820 thread(1) tag(0) WARN recv_forward_close_resp:2689 Unable to receive Forward Close response, PLCTAG_ERR_BAD_STATUS!
2025-01-16 18:08:09.820 thread(1) tag(0) WARN perform_forward_close:2338 Forward close response not received, PLCTAG_ERR_BAD_STATUS!
2025-01-16 18:08:09.820 thread(1) tag(0) WARN session_handler:1487 Forward close failed PLCTAG_ERR_BAD_STATUS!
2025-01-16 18:08:09.820 thread(2) tag(0) WARN check_read_request_status:1128 Session reported failure of request: PLCTAG_ERR_TOO_LARGE.
2025-01-16 18:08:09.820 thread(3) tag(11) WARN plc_tag_create_ex:1056 Error PLCTAG_ERR_TOO_LARGE while trying to create tag!
2025-01-16 18:08:09.820 thread(3) tag(11) WARN socket_write:1889 Socket pointer is null!
2025-01-16 18:08:09.820 thread(3) tag(11) WARN send_eip_request:2207 Error, -25, writing socket!

The message is received properly but the library is not able to process it. 

[image]

Is there any way to increase the buffer size?

jggri, 2025-01-17:
[image]

jggri, 2025-01-17:
https://github.com/dmroeder/pylogix/issues/124

jggri, 2025-01-17:
[image]

[image]

jggri, 2025-01-17:
> [kyle-github](https://github.com/kyle-github)
If I get a CIP error code 0x08 (Unsupported), then I drop back and retry using an old Forward Open on the second try. However, if I get a CIP error 0x01 then I dig a little deeper. If you get 0x01 as the main CIP error byte and there are two additional words (not bytes!) of status, then I read the next two words. The first is the additional status.
> 
> If that additional status word is 0x109, then the next word appears to be the size that the PLC will accept for a request packet. It matched 4002 exactly which seemed to be a bit of a coincidence.
> 
> So the bytes are:
> 
> 0xDB - Forward open extended response
> 0x00 - padding of some sort. Seems to be zero.
> 0x01 - error code 0x01
> 0x02 - two additional words of error status.
> 0x09
> 0x01 - error code 0x109
> 0xA2
> 0x0F - 4002 in hex
> 
> I have tried this against a ControlLogix using an ENBT network module. The CL was running firmware v15 and Forward Open Ex did not work and I got 0x08. I tried against a CL that was running v20 and it did not work. I tried another with v21 or 24 (I can't remember) and and EN2T which had the last/latest firmware at that time (circa v24 of the CPU firmware) and it did work to negotiate a 4002 packet. I tried with larger packet sizes and that is when I discovered the above error behavior. The PLC would respond with the error above but tell me that it wanted a 4002-byte payload. I tried with smaller numbers and it worked as well IIRC.
> 
> I have a chassis with a L55, L61 and L81, and a couple of ENBTs and it works with the following combinations:
> 
> L81/L81 network port - works fine, negotiates 4002 bytes.
> L81/ENBT - Does not work. I get 0x08 and negotiate to 500 or whatever it is bytes.
> L61/L81 network port - works fine, negotiates 4002 bytes.
> L61/ENBT - does not work, negotiates 500 bytes.
> L55/* - never works with Forward Open Ex and only supports 500 bytes.

Is there any way you check what was the connection size negotiated?

kyle-github, 2025-02-28:
Somehow I've missed this.   Huh.   I never put in a way to get the connection buffer size.   This should be both settable and gettable.  The reason for setting it is that in some circumstances, bridging over ControlNet IIRC, the max buffer is only 502 byte or something weird.

kyle-github, 2025-03-04:
Any update?  Please reopen if you have more information that you believe shows that the library is doing the wrong thing.  I am very interested to find out how you got that large response packet!

closed 2025-03-04 by kyle-github

kyle-github, 2025-04-11:
Note that Micro8x0 series PLCs can supposedly have a 64kB buffer.   But, they cannot support multi-request packets.

Any more information on what kind of PLC this is?   Version of firmware?

jggri, 2025-05-02:
It's not a real device. It's a server created using https://github.com/liftoff-sr/CIPster

kyle-github, 2025-05-02:
Ah, OK.   Then the code for the server is probably ignoring the handshake negotiating the buffer size.