libplctag/libplctag #515: Memory leak on connection loss / PLCTAG_ERR_TIMEOUT

HalfenD, 2025-01-20:
## Description
When reading a tag and a disconnect/**PLCTAG_ERR_TIMEOUT** occurs, there is a memory leak.  
I analyzed the error with `heaptrack` and found this line allocated the memory that leaked: https://github.com/libplctag/libplctag/blob/c55bc5876d938dda1c609750cde5ae4812d7b8a8/src/protocols/ab/session.c#L2778

## Steps to Reproduce:

Create a tag with for example with auto_sync_read. (manual read results in the same)
Disconnect the network between some request and now.
Observe the memory leak.

## Expected Behavior:
There should be no memory leak when a disconnect occurs during a read process.

## Environment:
Software Version: V2.6.3 and release-branch
PLC: ControlLogix EIP_1769
Operating System: Debian-Linux amd64

kyle-github, 2025-02-26:
I think I need to spend more time looking for more of these.

kyle-github, 2025-03-04:
This should be fixed with 2.6.4 but I need better tests.   If you see this again, please reopen this or open a new issue.  Again, thank you for finding and filing this!

closed 2025-03-04 by kyle-github