libplctag/libplctag #517: Force using Forcing the use MULTIPLE SERVICE PACKET 0x52

jggri, 2025-01-22:
I have a program that reads all the tags in a ControlLogix PLC every second. I am interested in maximizing its speed, so I need to use service 0x0A, which is the Multiple Service Packet where you can include several reads in the same request. There was an issue ([#15](https://github.com/libplctag/libplctag/issues/15)) where it was recommended to use async.c. 

Taking that file as a reference, in my application I see that the Multiple Service Packet service is used the second time the tag is read

[image]

The first time you read the tag it uses the service 0x52 which is 

[image]

Is there a way to force the use of Multiple Service Packets from the start, without waiting for the second read?

kyle-github, 2025-01-22:
It depends on timing internally.  If there are multiple requests lined up then the connection/IO thread for the PLC will pack them.  If the PLC supports packing.   If there is only one queued by the time the thread wakes up for whatever reason, it will use the single request form.   I will look at the code to double check that there is not a guard for the first read.

When you open a tag handle, the library does an internal read first to get the data type and to get the initial value.   This is so that the first operation the app can do is a write.   For instance if you have a tag that is a chunk of an array and you want to write just one element, you need to either make that one element a new tag handle or have an initial read to get all the other values.

jggri, 2025-01-22:
What you mentioned is exactly what was happening. A read was performed when creating the tag. In this case, I wasn’t using 0 for the timeout, so the reads were performed one by one.

[image]

In this case, I use the library to perform reads, so I’m not sure if there is a way to avoid that initial read using a parameter or configuration

kyle-github, 2025-01-23:
There is no flag, but that is an interesting idea if you know that your tag is going to be read only!

joed74, 2025-01-28:
I have two tags which i read with auto_sync_read_ms every 250ms and two raw tags (attribute lists) which i read with auto_sync_write_ms every 1500ms .. but i get mixed results:
[image]

Sometimes "multiple service packet" reads, sometimes "read tag fragmented" reads... 
Would be nice if i could group tags together for a "multiple service packet"-read ...

kyle-github, 2025-01-28:
Based on the screen shot, it looks like timing.   If the first non-packed request was the only one in the queue at the time the OS woke up the PLC/IO thread, then it will use the more efficient non-packed command.  During the time that the thread is waiting for a response, the next tag may trigger a read.  The closest I see these requests is about 4ms.  At least on my wired network that is longer than the response time from a ControlLogix PLC.

If you had a higher load with a lot more tags, you would see more packing.   

The library code has an async hand off between the application thread calling the API functions and the background PLC/IO connection thread.   When a request is queued, the application thread wakes up the PLC/IO thread.  However, the OS can take a while to actually start the thread.  I have seen up to 20ms on Windows.  Usually lower, but sometimes it is not.   

It looks like the library is working as expected.

kyle-github, 2025-02-25:
Please reopen if you think that the library is doing something incorrectly.

closed 2025-02-25 by kyle-github