libplctag/libplctag #518: Service 0x4C Read Tag Service and Symbol Instance Addressing

jggri, 2025-01-30:
In Logix 5000 Controllers Data Access, it is specified that there are two types of services for reading:

- **Read Tag Service (0x4C)**
- Read Tag Fragmented Service (0x52)

The function int plc_tag_read(int32_t tag, int timeout); always uses 0x52.
Is there any way to configure it to use 0x4C instead?

Additionally, the documentation mentions that the first four services preceding can be used with two addressing methods:

- Symbolic Segment Addressing
- **Symbol Instance Addressing (available in version 21 and later).**

Is Instance Addressing supported?

kyle-github, 2025-01-30:
Instance addressing is not yet supported.   One challenge is that the PLC responds differently sometimes when using instance addressing or symbolic addressing.

There is not a direct attribute to control use of fragmentation.  Right now the library defaults to using it if the PLC type supports it.  Not a bad idea to make it an attribute.

kyle-github, 2025-03-04:
This is a duplicate of part of #453.

closed 2025-03-04 by kyle-github