libplctag/libplctag #526: cmake build for v2.6.4 doesn't install libraries.

daeho-ro, 2025-04-15:
Hello, 

I'v a problem that the build for v2.6.4 doesn't install libraries and .pc file.
So, I applied a patch by myself from the file `src/libplctag/CMakeLists.txt`

1. `# install(` -> `install(`
2. `if(EXISTS "${CMAKE_LIBRARY_OUTPUT_DIRECTORY}/libplctag.pc")` -> `if(TRUE)`

But I don't know whether this is a correct way or not. There should be a better way.

Thanks,

Related PR :
- https://github.com/Homebrew/homebrew-core/pull/219336

kyle-github, 2025-04-15:
Hmm, it looks like I managed to to overwrite the wrong diff.   There were over a hundred different commits to merge into HEAD and it looks like I did it incorrectly at least there.  I'll look into this and fix for 2.6.5.

Thanks for bringing this up!

kyle-github, 2025-04-30:
I just released 2.6.5.  Please let me know if I corrected the install fixes.

closed 2025-04-30 by kyle-github