libplctag/libplctag #531: Unable to Create Tag Using CIP communication Path with "1,6,2,192,168,100,102,1,0"

nazirhussain05, 2025-05-13:
Hi,

I'm trying to use libplctag to create a tag with a communication path like this: 1,6,2,192,168,100,102,1,0

But when I try to use this path in the tag string, I get an error and the tag fails to be created. **PLCTAG_ERR_BAD_GATEWAY** :

Any help in understanding what I might be missing or how to fix this path would be appreciated!

Thanks.

kyle-github, 2025-05-14:
The path is not correct if you are trying to bridge through a network module.   Guessing from what you have, I think the correct path would be:

`1,6,18,192.168.100.102,1,0`

Please reopen if that does not work for you!

closed 2025-05-14 by kyle-github

nazirhussain05, 2025-05-15:
Thanks for your help.

From the code comments, I understand that:
-   18 is used for routing via port 2 / port A 
-   19 is used for routing via port 3 / port B 

But I'm wondering:
When would I need to use 19 instead of 18? as I receive a CIP path like((1,6,2,192,168,100,102,1,0) this dynamically from a configuration:

Any guidance on when and why to use 19 would be appreciated, especially for dynamically handling different topologies in code.
Thanks again for your great library and support!

kyle-github, 2025-05-15:
Yes, 18 is network module port "A" and 19 is network module port "B".   There are not that many modules that have a second port.  If you are using an ENBT or EN2T there is only port "A".  

CIP paths are composed of pairs of [out port, node ID/address].
```
1,6 -> [port 1 (backplane), node ID 6 (slot 6 on the backplane)]
2, 192.168.100.102 -> [port 2 ("A"), IP address 192.168.100.102]
1,0 -> [ port 1 (backplane on the remove PLC), slot 0 (probably the CPU module)]
```
That configuration is missing important data.   You would need to know that slot 6 holds a network module in order to know that the "2" means the "A" port.  And then the IP address just happens to look like an IP address because it uses the 192.168.x.y class B non-routable address block.  Maybe you could use AI to figure it out, but in simple code?  Is there any way you can get more information about the configuration?