libplctag/libplctag #532: Error trying to read a CompactLogix Alias tag

themouseNZ, 2025-05-14:
I need to read a tag that is an alias from a function block. The old original Citect HMI accessed the tag no issue, but when I try with libplctag it fails. My understanding is an alias is just a pointer in the controller memory so not sure why is breaking. The function block has alarm and status bits that are combined into alias words.

[image]

I am trying to read .ALRM and .STAT alias

[Single_Alias_Tag_Create_Error.txt](https://github.com/user-attachments/files/20212249/Single_Alias_Tag_Create_Error.txt)

kyle-github, 2025-05-14:
Thanks for using libplctag and providing a log file!

The log shows that you are getting the tag `ER1_CND1_M001.ALRM`.   Does that sound right?   It is succeeding and returning a DINT.  Did you attach the wrong log file?  Can you provide a failing tag attribute string?

themouseNZ, 2025-05-14:
Hi, I love libplctag. Am using it for several old slc500's but now need to use on compactlogix and having a few challenges!
This Is a log trying to read several of them :-

[Alias_Tag_Create_Error.txt](https://github.com/user-attachments/files/20212592/Alias_Tag_Create_Error.txt)

themouseNZ, 2025-05-14:
I get Code -19 - PLCTAG_ERR_NOT_FOUND Tag when trying to access these

kyle-github, 2025-05-15:
It looks like it works find until you try to get `ER1_CND2_M001.ALRM`.  Then the PLC sends back an error that it cannot find the tag.

Can you try the tool `list_tags_logix`?  Use it to see what tags the PLC thinks it has.

I am not sure how the tags you are reading match to the screen shot of tags in the first post above.   Is that post of a UDT and all its fields?   Are these program tags vs. controller tags?

themouseNZ, 2025-05-15:
Thank you for looking for me. Much appreciated :-)

closed 2025-05-15 by themouseNZ

kyle-github, 2025-05-15:
Please reopen if you find an issue with aliases!   They should work, but "should" is not "will"!