libplctag/libplctag #536: auto sync tags do not continue after remote goes down and comes back

kyle-github, 2025-06-01:
Found by @wegel.  See PR #530.

When using auto sync tags, the auto sync stops after remote goes down and comes back.

Fixed in PR #530 and PR #535.

mikecflood, 2025-06-01:
This does not just apply to auto sync tags.

If one uses plc_read_tag(libplctag_id, 0) and the remote connection goes down, it will return PLCTAG_ERR_BUSY. It will continue to do this even after the remote connection comes back up.

The only way to get passed this is to call plc_tag_destroy(libplctag_id) for every tag that uses the remote connection and then re-open each one.

kyle-github, 2025-06-02:
Thanks for pointing this out!  I will make some tests and fix this too.

kyle-github, 2025-06-15:
fixed in 2.6.6

closed 2025-06-15 by kyle-github