libplctag/libplctag #558: in certain condition, payload is over 4002 byte and go into reconnection loop

byunghyunna2, 2025-07-15:
I had some issue that after hours run, lost communication to PLC and not being able to recover back.
I ran wireshark and debug option 2, noticed libplctag sent 4003 byte of payload then no response from PLC when into reconnection.
it seems payload count was wrong and overwrite a byte to hostIP address, so it went into reconnection loop due to incorrect ip address.

https://github.com/libplctag/libplctag/blob/86af977d54f2bc9cffd77b4c3d185642c48d7d28/src/libplctag/protocols/ab/session.c#L1605

maybe it should account for size of sequence count?

I'm using 2.6.6 on windows 11 and CompactLogix 5069-L306ER

[image]

[stderr.log](https://github.com/user-attachments/files/21241859/stderr.log)

kyle-github, 2025-07-17:
Yeah, it is pretty clearly an overwrite.  I will put out a patch in a couple of days to make sure this does not cause overwrite but to really fix it is going to take a bit longer as there is clearly something in the logic that is calculating the available space incorrectly.

Thanks for finding this!

kyle-github, 2025-07-23:
Please reopen if this is still unfixed.  Thanks again for bringing this up!

closed 2025-07-23 by kyle-github