libplctag/libplctag #564: Does connecting an Omron PLC via EtherNet/IP support implicit IO messaging?

alexhaifeng, 2025-07-24:


alexhaifeng, 2025-07-24:
Each TCP message requires an acknowledgment, but if UDP is supported, the latency would be lower.

kyle-github, 2025-07-25:
There is no support for UDP/implicit messaging in libplctag at this time.  The existing API is not that well suited for it since it relies on a request/response pattern.   Doing device discovery or using multicast and getting produced tag data would be really nice to have, but again it is a large addition/change to the existing API and I have not found a good way to include that without just adding completely new calls.  If you have any ideas what the API could look like, I would love to hear them!