libplctag/libplctag #565: Possible Regression error: -34 (PLCTAG_ERR_TOO_SMALL)

mikecflood, 2025-07-27:
Creating the tag is successful however when I try to write to it, I am getting an error: -34 (PLCTAG_ERR_TOO_SMALL

The tag being used is: protocol=ab_eip&plc=controllogix&gateway=dns_name&path=1,0&name=TAG[0].15&elem_size=2&elem_count=1

This worked with all versions up until v2.6.7, it stopped working with v2.6.8

kyle-github, 2025-07-27:
What kind of array is it?   Is it an INT array?  Thanks for bringing this up!

mikecflood, 2025-07-27:
Yes it is an INT.

kyle-github, 2025-08-01:
Should be fixed in 2.6.9.

closed 2025-08-01 by kyle-github

mikecflood, 2025-08-16:
While this could be a new issue, I think this same issue exist for the micrologix 1400. Now that issue #570 has been fixed I was able to create and read tags. However I was not able to write individual bits.

The tag I was using is:

protocol=ab_eip&plc=micrologix&gateway=[DNS_NAME]&name=B9:0/0&elem_size=2&elem_count=1

This is the level 4 debug: [debug_level_4.txt](https://github.com/user-attachments/files/21817799/debug_level_4.txt)

If it helps I was able to write an int, this is the tag used to write to an int which worked:

protocol=ab_eip&plc=micrologix&gateway=[DNS_NAME]&name=N11:0&elem_size=2&elem_count=1