libplctag/libplctag #570: Micrologix 1400 PLCTAG_ERR_REMOTE_ERR(-29)

mikecflood, 2025-08-05:
Creating tags is successful. Reading the tag after it is created fails with PLCTAG_ERR_REMOTE_ERR (-29).

I tested release going backwards and found this first starting occurring in v2.6.8 (works without issue on v2.6.7).

The tags being used are:

protocol=ab_eip&plc=micrologix&gateway=dns_name&name=B9:0&elem_size=2&elem_count=7
protocol=ab_eip&plc=micrologix&gateway=dns_name&name=B10:0&elem_size=2&elem_count=7
protocol=ab_eip&plc=micrologix&gateway=dns_name&name=B18:0&elem_size=2&elem_count=2
protocol=ab_eip&plc=micrologix&gateway=dns_name&name=N11:0&elem_size=2&elem_count=2
protocol=ab_eip&plc=micrologix&gateway=dns_name&name=N12:0&elem_size=2&elem_count=33
protocol=ab_eip&plc=micrologix&gateway=dns_name&name=L32:0&elem_size=4&elem_count=6

Note: I use micrologix, compactlogix and modbus. This only occurs with the micrologix.

kyle-github, 2025-08-07:
Can you capture the debug output?   Set the debug level to 4.   I have a Micrologix, so I can probably try some of this.   Getting my copy of RSLogix 500 to actually work in a VM...  Sometimes it stays up as long as five minutes!

Thanks for bringing this up.  Are you trying with 2.6.9?

kyle-github, 2025-08-14:
This should be fixed with v2.6.10.  I tested with my Micrologix and set up more tests to make sure that this does not regress.  Thanks again for raising this issue!

closed 2025-08-14 by kyle-github