libplctag/libplctag #572: Can not write individual bits to micrologix 1400

mikecflood, 2025-08-19:
I think this is the same issue as #565 but for the micrologix 1400. Now that issue https://github.com/libplctag/libplctag/issues/570 has been fixed I was able to create and read tags for the micrologic 1400. However I was not able to write individual bits.

The tag I was using is:

protocol=ab_eip&plc=micrologix&gateway=[DNS_NAME]&name=B9:0/0&elem_size=2&elem_count=1

This is the level 4 debug: [debug_level_4.txt](https://github.com/user-attachments/files/21860054/debug_level_4.txt)

If it helps I was able to write an int, this is the tag used to write to an int which worked:

protocol=ab_eip&plc=micrologix&gateway=[DNS_NAME]&name=N11:0&elem_size=2&elem_count=1

kyle-github, 2025-08-20:
Sorry for the delay.  This one confused me for a long time.  The problem seems to be that you have a "elem_size" attribute.  There is some interaction here that I need to trace down.  Here is what I get when I leave that out (this is against a real Micrologix):

```
./tag_rw2 --type=bit '--tag=protocol=ab-eip&gateway=10.206.1.36&plc=micrologix&elem_count=4&name=B3:0/6' --write=1
Library version 2.6.10.
Processing argument 1 "--type=bit".
Processing argument 2 "--tag=protocol=ab-eip&gateway=10.206.1.36&plc=micrologix&elem_count=4&name=B3:0/6".
Processing argument 3 "--write=1".
Number of write args 1.
bit 6 = 0
New values written to tag.
bit 6 = 1
```

The `elem_count` attribute is ignored, but it looks like `elem_size` is not ignored, or is only partially ignored.

Trying leaving the `elem_size` attribute out of your tag strings and see if that works.

I am going to call this a bug for now because a bit tag should ignore the element size for a PCCC/DF1 PLC.

Thanks for bringing this up!

mikecflood, 2025-08-25:
When I checked out the prerelease branch it only had the version bump. Seems the source changes are not in it so wasn't able to test it.

mikecflood, 2025-08-26:
If you put any changes on the prerelease branch I am more than will to help test it out.

mikecflood, 2025-09-04:
I tested this with my setup and all read/writes work. Again, no DH+ with my setup.